sebanc/brunch #902: samus updates after july?

zainullahk, 2021-02-22:
after june/july updates wont come what image should i use?

sebanc, 2021-02-22:
In theory, intel gen < 4 would have to use LaCrOS from samus recovery image when samus is AUE (usually Google provide updates for up to 6 months / 1 year after the AUE theoretical date if I remember well).

In practice, for some reason, it has now been 2 releases that rammus recovery image is again compatible with older intel cpu. However, I am no sure if this will last.

zainullahk, 2021-02-22:
So I won't get chromeos updates and there isn't any other image  for 3rd
gen?

sebanc, 2021-02-22:
No, all chromebooks with such cpu will be AUE.

zainullahk, 2021-02-23:
Ok, thanks for the clarification.

zainullahk, 2021-02-23:
but i have another issue my camera stopped working it used to work in
chromeos 86
and chromeos is really slow compared to 86