sebanc/brunch #906: Webcam, Mic, and Sleep doesn't work!

pakdhealif, 2021-02-23:
Hello @sebanc,
I've just installed chromeOS 88 samus with brunch 88 latest on my ThinkPad X230 (i5 3rd gen, 8GB RAM)

Everything works fine except Webcam, Mic, and Sleep.

Here's what I've done:

- Microphone: this is the weird one. Google Meet or even Duo can't detect my camera and mic, but Google Assistant recognizes my mic fine.
- Webcam: God, I have no idea. Is it because my chromeOS version is the latest? I heard on issue #691 that downgrading to lower version could be the solution, but is there any resources on how can I downgrade? Note: Device ID is 04f2:b44d, made by Chicony, works fine on other Linux distros.
- Sleep: activated s3 and acpi power via brunch-toolkit, didn't work.

Any suggestions will be appreciated. Thanks!

sebanc, 2021-02-24:
For mic, this is a weird issue indeed, you could try "internal_mic_fix" option but I am not sure if it will help.

As far as I can tell, your camera should work, maybe try kernel 5.10.

Suspend is hit or miss with the linux kernel, maybe kernel 5.10 might also help but those issues are difficult to debug.

khanhtranngoccva, 2021-03-01:
I prefer the "octopus" or "hatch" image. Do get them.