sebanc/brunch #907: Which latest Brunch is compatible for my PC?

Sayor0630, 2021-02-23:
Which latest version of brunch I can use without any problem? Here is my PC configuration:-
[image]

khanhtranngoccva, 2021-02-24:
Use the rammus/hatch system image. Just check if you have integrated graphics by entering "devmgmt" in the Start menu and look for the display adapters section. If you have any sort of HD/UHD/Intel IRIS graphics then you can run it.

Sayor0630, 2021-02-24:
> Use the rammus/hatch system image. Just check if you have integrated graphics by entering "devmgmt" in the Start menu and look for the display adapters section. If you have any sort of HD/UHD/Intel IRIS graphics then you can run it.
Oh. I have Intel HD 4600

Sayor0630, 2021-02-24:
Can I use latest **Brunch r88 stable 20210223** on my **PC Config?**

Sayor0630, 2021-02-24:
I can't use Kernel 5.4 based brunches. If I want to use K5.4 or higher, my PC just Reboots after Brunch setup. But if I use Kernel 4.19 based Brunches, it reboots perfectly. I use use the latest **Brunch r88 stable 20210223**? You are saying that **Improve support for some old intel GPU on kernel 5.4**. Can I use this release now?

sebanc, 2021-02-24:
This build is indeed supposed to fix issues like yours but it is the first time I hear about such issue on a 4th gen intel. The best way to know is to try it.

Sayor0630, 2021-02-24:
But if I try, will it boot?

Sayor0630, 2021-02-24:
ooh. I am using triple boot.

sebanc, 2021-02-24:
We won't know unless you try, maybe the bug I fixed was related to the issue you had or maybe it is something else.

Sayor0630, 2021-02-24:
I have updated to kernel 5.10. It is working. But I can't run using Kernel 5.4. By Kernel 5.4, ChromeOS doesn't boot. After the rebuilding and brunch loading, PC just restarts everytime!

Sayor0630, 2021-02-24:
[image]
Is this latest version of rammus chromeos?

sebanc, 2021-02-24:
That's the latest recovery image version available but you can enable updates if you want to have r88 (see the main readme).

In order to debug your issue with kernel 5.4, I would need a short video of the boot process in debug mode (press "e" when you see grub and remove "console= vt.global_cursor_default=0 brunch_bootsplash=default quiet", then F10 to boot)

ameyjoshi3068, 2021-02-25:
I also have the same issue with 5.4 and haven't fixed with the latest update. 
Should I upload a video of my boot here?

ameyjoshi3068, 2021-02-25:
@Sayor0630  also remove `/`  in the ending of the previous line of `console= vt.global_cursor_default=0 brunch_bootsplash=default quiet` before recording video.