sebanc/brunch #911: Can this PC run your brunch with chrome os?

Sayor0630, 2021-02-24:
[image]

DerFeineOtter, 2021-02-24:
Try this to find out: https://github.com/WesBosch/brunch-toolkit 
you have to run it from Linux so use a live USB.

Sayor0630, 2021-02-24:
Pls anyone reply!

khanhtranngoccva, 2021-02-25:
Appears not. Sorry.

khanhtranngoccva, 2021-02-25:
The following processors are not currently supported by Brunch, and it's unlikely that any support will come for these CPUs:

All ARM processors. This includes rockchip and mediatek
All Intel Netburst variants.
Any Intel Atom, Celeron or Pentium variants made BEFORE 2012.
All Intel Core Solo and Core Duo variants, including Core 2 Duo.

Well you can still try it, but it's kinda unlikely to work.

princepjamie06832, 2021-02-25:
Unfortunately, using dGPU or onboard may cause of rendering unbootable. Core 2 variants cannot run due to compatibility issues.

Sayor0630, 2021-02-25:
ooh. Then it can't.  can the 1st version of brunch support?

ameyjoshi3068, 2021-02-25:
@Sayor0630  I would suggest you try installing brunch in this PC.
If it worked then well and good. 
You must be having all the necessary files in another PC.

Sayor0630, 2021-02-25:
> @Sayor0630 I would suggest you try installing brunch in this PC.
> If it worked then well and good.
> You must be having all the necessary files in another PC. You can use that also.

ok

WesBosch, 2021-03-04:
Brunch is not compatible with this CPU at all, and it will not work.