sebanc/brunch #912: Camera Not Working Correctly

DerFeineOtter, 2021-02-24:
So i use the latest Chrome OS rammus and Brunch on my Chuwi UBook 11.6. Everything works except my front Camera.
My Camera on the back works, but my front Camera just loads for 2 seconds and then switches back. Also if i try to use Zoom it says my Cam is used by another application.

Lucaacer, 2021-02-24:
Try to delete /etc/camera folder and reboot.

It worked for me.

DerFeineOtter, 2021-02-26:
thx it worked

DerFeineOtter, 2021-02-26:
but now my android apps wont work

Lucaacer, 2021-02-26:
> but now my android apps wont work

I am sorry, but the 2 issues are not related to each other.

Ebbot9876, 2021-02-26:
I also tried to delete /etc/camera, android apps stopped working!
Chuwi HI 10X

Lucaacer, 2021-02-26:
It might have got something to do with rebuilding the framework since removing that folder can't conflict with the android container and, by the way, that suggestion comes from a post I had tha chance to read here some time ago.

DerFeineOtter, 2021-02-26:
so can u rebuild the framework manualy ? i try it rightnow with updating both ChromeOS and Brunch via brunch toolkit

DerFeineOtter, 2021-02-26:
yeah it like before deleting /etc/camera so what about clearing the folder and giving it a sort of writeprotection so nothing will change ?

Lucaacer, 2021-02-27:
I just meant that, whenever you modify something and the OS reboots, the framework gets rebuilt. I frankly have no idea why removing that folder might prevent the android container from working, you'd better ask Sebanc, sorry.

sebanc, 2021-02-28:
Could you try the new brunch-unstable build with "options=no_camera_config" added on the kernel command line ?
https://github.com/sebanc/brunch-unstable/releases

DerFeineOtter, 2021-02-28:
yes of course my lord! and btw, normal cam works with deleting some files of camera folder instead of deleting the complete one. Can i update using brunch toolkit ?

DerFeineOtter, 2021-02-28:
now it uses the wromg camera in zoom etc...

DerFeineOtter, 2021-03-01:
Update : web version of apps work but not android apps

tailz303, 2022-10-11:
> Try to delete /etc/camera folder and reboot.
> 
> It worked for me.

This entirely broke all apps for me and also broke both WiFi and Bluetooth. Had to reinstall. I would recommend going another route, however according to https://developer.android.com/topic/arc/manifest it looks like camera along with many other API calls are just not supported between chrome OS and android apps. Sad but its what google has done.

thomasjungblut, 2022-10-25:
hey folks,

I'm on rammus (v100 due to #1546), running on a surface go 1. It works really great and gives the hardware extra lifetime, so thank you @sebanc :)

I also figured the camera doesn't get recognized. I'm on the 5.15 kernel already, the no_camera_config toggle has no effect.

I've found 
https://github.com/linux-surface/linux-surface/wiki/Camera-Support

So I would love to apply the kernel patch and try the libcamera approach. Is there an easy way for me to load an alternative kernel into brunch?

dmitrmax, 2025-12-16:
It's been almost six years for this issue and still no progress?

I've installed the latest ChromeOS 142 (octopus) on my kid's Acer TravelMate B118-M. Everything's fine except that Microsoft Teams (as an Android app) doesn't  use the camera. I'm not sure if Teams detects the camera at all. Google Chat to the opposite uses camera just fine.

From this I also made a conclusion that Android emulation layer has some issues with camera. How can I diagnose the root of the problem an maybe help to resolve the issue?