sebanc/brunch #915: Black screen after update to the last release

JamesNoone69, 2021-02-25:
Hello! 

I updated to Brunch r88 stable 20210223 release. Now, I have a black screen after a few minutes using Chrome or any app. I can only move the cursor of the trackpad but the screen is full black. I have to force stop the computer. I downgraded to previous version.

I use this with version 87 of rammus recovery image and version 88.0.4324.186 of ChromeOS.

Thanks!

sebanc, 2021-02-25:
Hello,

What are your pc cpu/gpu ?

JamesNoone69, 2021-02-25:
> Hello,
> 
> What are your pc cpu/gpu ?

It's an Intel Core i5-4200U

lfom, 2021-02-26:
Same problem here:
1. I had Brunch with ChromeOS 86 installed on an image in a partition of my internal disk
2. Downloaded and updated Brunch to latest release (20210223)
3. Installed and rebooted, updated partition fine, no errors
4. Went to Settings and updated ChromeOS
5. Upon restart it showed an errors, something like "ChromeOS was not found in any device" and loaded a shell

I still can access and mount the image from Linux, the only thing weird I noticed is that it does not have a ROOT-B partition, only A, A1 and C.

bobafetthotmail, 2021-02-28:
I have the same problem on my  i3-4020Y,  Dell Venue 11 7130 and **Brunch r88 stable 20210223** (latest stable release at at the moment)

I also saw visible graphical artifacts for the short time the screen works after a reboot.

gustavofalcao1, 2021-02-28:
Same probleme i5-5200U

gustavofalcao1, 2021-02-28:
In my case, a black screen with a blinking cursor was appearing. And when I tried UEFI mode, I didn't find the SSD. 
So I did the installation in UEFI and started the boot with USB with Ubuntu, I selected the option "boot with next media" or something ... 
Then my SSD appeared and I selected it and then ChromeOS started!

I hope I can help someone else!

Garykom, 2021-03-08:
Also black screen after installation on Acer c720p
Rammus 87, Branch 88

Firnin, 2021-03-11:
Same problem for me, also on Acer C720P, rammus, after updating Brunch from r87 to the latest r88 build.
Blackscreen almost instantly after login, weird graphic glitches and artifacts on login screen. I can still see and move my cursor after the screen turns black, and the DEV console (ctrl + alt + f2) works just fine

wanglrebe, 2021-03-13:
i3 4gen too

b1t0y, 2021-03-13:
same here with i5 4gen haswell with r88 brunch build.  Any solution?

lfom, 2021-03-13:
Have you guys tried to use kernel-5.10 instead of the default kernel?

JamesNoone69, 2021-03-13:
> Have you guys tried to use kernel-5.10 instead of the default kernel?

Hi ! I tried 4.19 but I had to powerwash then, so I'm afraid it happens again with 5.10 :)

Firnin, 2021-03-13:
Ok, so I changed the Kernel to 4.19, did a Powerwash and so far now black screen!
I tried 5.10 first, but then I had a white screen after the Chrome OS logo disappeared.

nalmada, 2021-03-14:
Can confirm that swithcing to 4.19 and powerwashing fix the issue

salvavic, 2021-03-15:
Surface Pro 3 (i5-4300U) here. The issue occurs after I updated to r89. I switched to kernel 5.10 and fix this issue.

WiseMetalhead, 2021-03-15:
Installing the 5.10 kernel solved the black screen issue. However, graphic artifacts remain. It is simply impossible to use a laptop with them. I used Brunch r88 unstable 20210302 and rammus recovery