sebanc/brunch #916: About Drives in ChromeOS

Sayor0630, 2021-02-25:
I can't see any **Local Disks or Drive** here (except **Pendrive**) in my Chrome OS. How can I enable **Local Disks and Drives**?

ameyjoshi3068, 2021-02-26:
Use `mount_internal_drives` option to do so(See Readme).
But this will also make chrome os a bit slower.

Sayor0630, 2021-02-26:
ok

Sayor0630, 2021-02-26:
I have enabled it in the grub. But I can't see my drives! What can I do now?

Sayor0630, 2021-02-26:
Is my grub editing is right?
[image]

ameyjoshi3068, 2021-02-26:
No your options are not right
Wait..

ameyjoshi3068, 2021-02-26:
Let me know if it works

Sayor0630, 2021-02-26:
I will join discord. Can you give me invite link?

Sayor0630, 2021-02-26:
Is it right now?
[image]

ameyjoshi3068, 2021-02-26:
It looks OK try booting with it. You will get discord link in Readme

Sayor0630, 2021-02-26:
There is no Drives. It didn't work
[image]

Sayor0630, 2021-02-26:
ooh. It connected. Thank you!
[image]

Sayor0630, 2021-02-26:
But why it is read-only?
[image]

Sayor0630, 2021-02-26:
Play store and Play store apps isn't opening after I enabled Drives. What can I do now?
[image]

ameyjoshi3068, 2021-02-26:
> But why it is read-only?
> [image]

The partitions you are seeing in this images are not read only.
Try opening any partition. You will be able to write in it.

ameyjoshi3068, 2021-02-26:
About play store and android apps, try waiting 1-2 min at login screen.
According to my thinking, Mount internal drive stressed chrome os a little. So it needs time to load android container.

Sayor0630, 2021-02-26:
> About play store and android apps, try waiting 1-2 min at login screen.
> According to my thinking, Mount internal drive stressed chrome os a little. So it needs time to load android container.

ooh

bigcatnerd4159, 2023-10-21:
[image]
Well I have also got that read-only issue, any suggestions?