sebanc/brunch #917: 20210223 + kernel-5.10 = ChromeOS not found

lfom, 2021-02-26:
Hello!

ChromeOS does not load anymore, it stops before loading the splash screen. I have Brunch installed since some time ago, all updates worked fine so far. I was using kernel-5.10.

I had Brunch with ChromeOS v86 installed on an image in a partition of my internal disk
Downloaded and updated Brunch to latest release (20210223)
Installed and rebooted, updated partition fine, no errors
I checked Brunch version and it showed 20210223
Went to Settings and updated ChromeOS
Upon restart it showed an error saying something like "ChromeOS was not found in any device" and loaded a shell
I still can access and mount the image from Linux, the only thing weird I noticed is that it does not have a ROOT-B partition, but two A and one C. Actually I have noticed it some time ago, thought it was weird and that I would report it, but didn't do it in the end.
Intel Atom X5 convertible

Update: I have edited GRUB and changed it to use the default kernel, and it did the Brunch update stuff and it booted fine again. I was able to confirm it is running ChromeOS v88 and kernel 5.4.88. I will try to go back to 5.10 now.

Jerit3787, 2021-02-28:
Did it solve your problem?

lfom, 2021-02-28:
@Jerit3787 Yeah, as I reported above: first reverted to default kernel (5.4) and it installed Brunch after the update. Then went back to kernel 5.10, it did his thing again, everything seemed fine.

Jerit3787, 2021-02-28:
alright so please consider closing this thread. thank you

closed 2026-03-11 by github-actions[bot]