sebanc/brunch #919: What is rmmod tpm?

Sayor0630, 2021-02-26:


ameyjoshi3068, 2021-02-26:
Adding this line is necessary if you are booting through Ubuntu or Linux Mint grub.

midi1996, 2021-02-26:
It will remove tpm module from running (i think grub tpm module).

omaemae, 2021-02-28:
> Adding this line is necessary if you are booting through Ubuntu or Linux Mint grub.

@ameyjoshi3068 Assuming things isn't good. That one line causes GRUB on Arch Linux to throw a warning and pause booting for 5 seconds with
```
error: module not found
Press any key to continue booting...
```
(loosely translated from Polish)

You should detect if the current GRUB install has the `tpm` module, and add the `rmmod tpm` appropriately. Also, **don't** detect this based on distro. There may be some obscure distro which is based on Ubuntu which also has the TPM module, but isn't detected.