sebanc/brunch #925: cros_debug (space) options=broadcom_wl (space) loop.max

mikaolee, 2021-02-28:
Thinkpad E431 notebook, win10 dual system boot, Chrome OS system Wi-Fi button cannot be used, according to the issues prompt, in win10 system, open grub2win customer options, add "cros_debug (space) options=broadcom_wl (space) loop.max "Option, restart, enter the Chrome OS system, the Wi-Fi button is still gray, and wireless is still not applicable. Please help me, thanks.

mikaolee, 2021-02-28:
After re-operation, the wifi button can be turned on, but no network can be searched. If you manually add a wireless network, it prompts that it cannot be found. Wi-Fi is still not available. A newly discovered problem is that Chrome OS will not sleep after being idle for a few minutes, but will shut down directly. Why? Ask for help, thanks

mikaolee, 2021-02-28:
Manually connect to the wireless network, prompting that it is not within range. I wondered if the hotkey was off, and tried to open it with Fn+F8 or F8, which conflicted with the hotkey of the chrome system.

lucasgabrielmoreno, 2021-07-23:
> Manually connect to the wireless network, prompting that it is not within range. I wondered if the hotkey was off, and tried to open it with Fn+F8 or F8, which conflicted with the hotkey of the chrome system.

Did you solved it? How? Thanks!