sebanc/brunch #929: "Your system is repairing itself. Please wait." started showing after a few days.

m-obeid, 2021-03-01:
I am using a ThinkPad E580. I used the "rammus" image version 87. Downloaded from https://cros-updates-serving.appspot.com

It worked fine for a month or so but now it seems to show this screen for 5 minutes and it just restarts!

Same things happen and so am I stuck in this unexpected bootloop.

What can I do to fix it? Or is this a bug in brunch or something?

Sayor0630, 2021-03-03:
reinstall it. It got corrupted. Or you can try changing the kernel version

WesBosch, 2021-03-04:
If you're dualbooting, make sure windows was shut down fully by holding shift when clicking shut down. This issue is typically only seen in windows dualboot setups

m-obeid, 2021-03-07:
> If you're dualbooting, make sure windows was shut down fully by holding shift when clicking shut down. This issue is typically only seen in windows dualboot setups

Fixed the problem by turning off fast startup. Thank you!

closed 2025-05-04 by m-obeid