sebanc/brunch #931: Heating issue (Surface Pro 5 (2017))

nadlax, 2021-03-02:
I'm using a Surface Pro (2017) core i5, which is a fanless version. And just a few minutes of light usage seems to result in excessive heating to the point that it's not even viable to hold it. On a Surface Pro 4 i7 that I also own, the fans seem to run all the time, indicating very high cpu usage in both cases

saturn-hh, 2021-04-05:
I'm using the same Hardware with no problems. Can you give more details:
Which version of recovery  Image are you using? 
Have you changed the Brunch options? 
Which Kernel are using?

ouija, 2022-04-03:
I noticed this last night as well while running Brunch ChromeOS for the first time on a Surface Pro 4, fan was running constantly and heat was excessive, with major power drain _(as opposed to using FydeOS 14 for the Surface Pro 4)_

I was using `rammus stable 99.0.4844.57` recovery with default kernel (5.4), using `pwa`, `mount_internal_drives`, `ipts` no_camera_config` options

closed 2026-03-11 by github-actions[bot]