sebanc/brunch #932: Change Caps Lock key to Control key

bus710, 2021-03-02:
Hi, 

First, thanks a lot for this project. I can see that many people can re-purpose their old devices to be Chrome OS device - this is not just a technical project but a great rehabilitation for our planet. This deserves to be an official Google backed project in my opinion.

About 2 months ago, I tried Brunch on my XPS13 and everything worked fine for me. Just one thing I want to know is that, (in the settings) the Caps Lock key could only be changed as search, alt, or shift key - but no control key unlike other Chrome OS devices. Can we adjust this by any configuration? (Maybe it has been changed since I tried though).

Thanks again,

Sayor0630, 2021-03-03:
no

bus710, 2021-03-03:
I see. That is unfortunate but not a big deal as well.
I guess I should dive into the code by myself to find out about it.

closed 2021-03-03 by bus710