sebanc/brunch #933: Lenovo Miix700 cannot install Chrome OS

Hoking007, 2021-03-03:
1. Use brunch_r87_stable_20201227.tar.gz to install chrome_os_13505.111.0_rammus_recovery_stable-channel_mp-v2.bin to the U disk.
2. Boot into grub from the U disk, and see the ChromeOS option
3. After entering the ChromeOS option, Miix 700 has a black screen
4. This U disk can enter ChromeOS on other computers. Therefore, there should be no problem with this ChromeOS U disk.
Does it not support Miix700 installation? thank you very much!

propotato123, 2021-03-07:
Did you create the Chrome OS install on another computer? The image for chrome os may be not located on the USB drive. Do you have other OS installed on your laptop? Is secure Boot enabled? Can you post your grub boot config?

Hoking007, 2021-03-09:
I can enter Chrome OS on other computers, but cannot enter Chrome OS on Lenovo MIIX 700.
No other operating system is installed on my laptop. And I have turned off the secure boot.
It may be that my Lenovo MIIX 700 is too old. I have installed it as Fyde OS (A variant of Chrome OS). It is running well. Chrome OS is really amazing. Let us experience Chrome OS. It seems that I will not use Windows 10 anymore. .

Hoking007, 2021-03-12:
I seem to have found an abnormality. I tried GPT to burn the ubuntu mate 20.04 U disk, and then used the U disk to boot MIIX 700. I encountered the same problem as the ChromeOS U disk boot. After the GRUB boot menu was started, the screen was black and unable to boot. I doubt if MIIX 700 does not support GPT booting. Before I burned ubuntu mate 20.04 U disk with MBR, I can boot into it.

Hoking007, 2021-03-12:
I seem to have found the problem. I burned the ubuntu mate 20.04 U disk with GPT, and then used the U disk to boot MIIX 700. I encountered the same problem as the ChromeOS U disk boot. After the GRUB boot menu was started, the screen was black and unable to boot. Before that,I burned the ubuntu mate 20.04 U disk with MBR, which can be booted into. Therefore, I wonder if MIIX 700 does not support GPT startup

Hoking007, 2021-03-12:
> I have a different Miix device and my internal disk is recognized as /dev/mmcblk0, easy to mess up installations when using generic instructions which usually refer to /dev/sda or similar.

I have not burned into the system yet, I just inserted the USB flash drive with ChromeOS into MIIX 700, but MIIX 700 just went to the Grub menu, and then press Enter to enter the ChromeOS step, a black screen appeared, and then waited for a few hours without any response. I suspect it is because my MIIX 700 cannot start GPT+UEFI? 
My BIOS settings are as follows: 
1. Turn off secure boot; 
2. Boot mode: UEFI mode.

Zero-zer0, 2021-04-29:
Same issue with Lenovo Miix 700 (m3-6y30)

The test are above:
1. `Brunch r86 k5.4 stable 20201201` + `rammus 86`  --> OK
2. `Brunch r88 stable 20210223`         + `rammus 87` / ``   --> Cannot Boot
3.  Based on 1, upgrade brunch to `Brunch r89 stable 20210403` --> Cannot Boot
4. `Brunch r89 stable 20210403`         +  `rammus 89`/ `soraka 89` --> Cannot Boot
5. `Brunch r89 unstable 20210424`     +  `rammus 89`  --> Cannot Boot  

- The USB Drive and image works properly on another Kaby Lake-R laptop. 
- With brunch r88/r89, after choosing entry in grub, the screen went black or with a ‘-’ on the sche reen only. 
- Even in debugging mode, there's nothing on the screen.
- Tried to change kernel version to 4.19/ 5.10 in `Brunch r89`, still cannot boot. 

It seems that the brunch framework didn't start properly. If there needs further information, please let me know.

The grub config is below, which works in `r86`. The image is located at `/dev/sda4`.
```
menuentry "ChromeOS (boot from disk image)" {
	img_part=/dev/sda4
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug options=enable_updates loop.max_part=16 img_part=$img_part img_path=$img_path \
		console= vt.global_cursor_default=0 brunch_bootsplash=default
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

Zero-zer0, 2021-08-09:
The problem seemed to be solved in `r92-unstable-20210809`! And here is my current configuration for reference:

1. **Recovery Image** : `octopus 91` or `rammus 91`(use octopus with rotation sensor working).
2. `Secure Boot=on`.
3. Install from USB to SSD with **Single Boot** configuration. `Windows` is removed.
4. Don't forget to import Secure Boot certification first.

This issue could be tagged with solved.
Thanks to the devs for their effort on this awesome project!

Hoking007, 2021-08-10:
Thanks for your help, I will try to install according to your steps, thank you very much!

gustavofalcao1, 2021-09-25:
Hi. I managed to solve the problem of the disk not being recognized as a bootable disk. Using this guide:
https://itsfoss.com/no-bootable-device-found-ubuntu/

At the end of the boot creation described in the guide. I disabled boot security so the kernel could boot

closed 2022-02-02 by Hoking007