sebanc/brunch #934: Surface Go power button not working after waking up

bisoelpetjah, 2021-03-03:
Hi. Thanks to this wonderful project, I can enjoy a brand new experience on my Surface Go device. But there's one thing that annoys me.

When I enable `suspend_s3` option, the device suspends properly. It even wakes up without issues. Smart lock and instant tethering works well. But then the power button doesn't work at all. If I disable `suspend_s3` option, the power button works normally after waking up. But smart lock and instant tethering doesn't work.

I have tried enabling `acpi_power_button` option along with `suspend_s3`, but it doesn't work either. I've tried switching kernels between 5.4 and 5.10, even powerwashed a few times, but the problem persists.

[dmesg.txt](https://github.com/sebanc/brunch/files/6073755/dmesg.txt)

Thanks a lot! Please let me know if you need more information.

manelscout4life, 2021-03-05:
I have the same issue

tml89, 2021-04-22:
I have the same issue

closed 2026-03-11 by github-actions[bot]