sebanc/brunch #935: chromeos-install.sh file not running

Siphonophorae2, 2021-03-03:
When I run the install.sh file it runs well until it gets to running chromeos-install.sh, it reports an error:

chromeos-install.sh: line 7: syntax error near unexpected token `$'\r''

chromeos-install.sh: line 7: `usage()

My HP stream notebook has ubuntu installed and its internal memory is known as mmcblk0 which I have edited in the install.sh file. I have tried multiple versions of brunch and the rammus chromeos recovery but I haven't had any luck.

WesBosch, 2021-03-04:
We do not support that install.sh file and will not be providing help for it's problems. Try using the chromeos-install.sh as directed by the github readme and do not use video guides or external files that aren't linked here on this repo.

TheSwisscheese2013, 2021-03-08:
Did you ever figure it out?same issue here.

TheSwisscheese2013, 2021-03-08:
> When I run the install.sh file it runs well until it gets to running chromeos-install.sh, it reports an error:
>
> chromeos-install.sh: line 7: syntax error near unexpected token `$'\r'' [...]

Did you ever figure it out?

WesBosch, 2021-03-08:
> Did you ever figure it out?

Run the brunch install file (chromeos-install.sh) directly by following the github guide and don't use the install.sh file provided by kedar or other YouTube videos or blogs. 

This issue is probably because of a broken variable in that unsupported script, so following the correct guide will probably not have this problem

Siphowin, 2021-03-08:
Thank you so much for the help, I ran it directly from the instructions on github and it worked as I'm writing this on the laptop. The only reason I used the install.sh is that I had searched on google for putting chromeos on a laptop and it was those vids that came up, sorry about making the mistake but I'll know for next time!

closed 2021-03-08 by Siphowin