sebanc/brunch #937: Apps are continuously spinning

Nandha200, 2021-03-07:
Hi I am using the latest brunch and latest image ( 87) . Recently all my apps which I installed in playstore and continuously spinning . I did power wash then it worked fine but again the apps are continuously spinning please help me..

[image]

Sayor0630, 2021-03-07:
Are u using enable drives? It will slow down your chromeos a bit. If u can wait 2-3 mins at login screen, it will load faster. Or if u are not using enable drives, then wait for some time.it will load after some time. Its because HDD is slow. If u use SSD, it will load faster. Thank u

Lucaacer, 2021-03-07:
Please open crosh with ctrl + alt + t, the type shell and cat /var/log/messages | grep Android and let me see if your issue is tha same as mine https://github.com/sebanc/brunch/issues/842

Nandha200, 2021-03-08:
Nope I haven't enabled drivers

Nandha200, 2021-03-08:
this is what I got after I typed the code u said
2021-03-07T01:03:38.915583+00:00 INFO session_manager[1206]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2405
2021-03-07T01:07:06.631966+00:00 INFO session_manager[1206]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 2405
2021-03-07T06:10:47.225242+00:00 INFO session_manager[1206]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2405 stopped
2021-03-07T12:09:01.020626+05:30 INFO session_manager[1156]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 1948
2021-03-07T12:09:22.900143+05:30 INFO session_manager[1156]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 1948
2021-03-07T12:09:55.381788+05:30 INFO session_manager[1156]: [INFO:session_manager_impl.cc(1872)] Android container with PID 1948 stopped

Nandha200, 2021-03-08:
And by the way this problem will happen only sometime , I just want to correct this sometime issue

Lucaacer, 2021-03-08:
Thanks, it's a different issue. In your case, the android container starts and gets stopped, in mine it doesn't even start.

closed 2021-03-09 by Nandha200