sebanc/brunch #940: Why when I move

Andytheclown, 2021-03-08:


Andytheclown, 2021-03-08:
I wouldn’t say this is a issue - I just kinda wondered what it happens?
I’ve been testing the ChromeOS on various machine and found that I don’t have to reinstall if I move the HDD or SSD but when I do move, it’s like a new clean machine setup?
This isn’t a problem and probably a good idea, I just wondered why it happens and how it knows it’s a different machine.

Thankyou for what you’re doing with this project. It’s great fun

smartguy1196, 2021-06-12:
This isn't unique to Chrome OS. In fact Windows uses an explicit protection against doing this, which is why moving Windows hard drives can cause problems with booting. Every piece of hardware in your computer provides its own details to the running OS at run time and that is what is used to identify your PC. Some parts even have unique identifiers lac MAC addresses, so that if you were to reinstall a hard drive in an identical system, the OS can still detect it was a different machine. This is an antipiracy feature practised by most manufacturers.

- The usage of Github Issues is inappropriate for questions like these. Use QA sites like stackexchange or other forums to learn how computing works