sebanc/brunch #943: Touch pad not working properly

Nandha200, 2021-03-09:
Hi, I am using a EULA touchpad that is not functioning properly with Chrome OS. After many clicks only it's recognizing. This is not happening with my external mouse. Please help me out..

SL-Pirate, 2021-03-11:
does clicking gestures work?
Im so used to touchpad keys. Since the latest update 89 thouchpad keys barely works
power key also stopped working. Feeling a little bit laggy too

SL-Pirate, 2021-03-11:
I just found that if u click thouchpad button while you are touching only it works
try that out bro

potatocubez, 2021-03-12:
Same issue here plus scrolling with 2 finger gesture is inverted when swiping to the app drawer after the search/assistance tab

potatocubez, 2021-03-12:
Found a trick.
So you have to keep your finger on the touchpad while klicking 
That's all

SL-Pirate, 2021-03-12:
> Found a trick.
> So you have to keep your finger on the touchpad while clicking
> That's all

This is what exactly I told above

SL-Pirate, 2021-03-12:
> Same issue here plus scrolling with 2 finger gesture is inverted when swiping to the app drawer after the search/assistance tab

I think that is normal bro
Get ised to it