sebanc/brunch #947: Accidentally Updated to Chrome OS 89. Trackpad Button Sometimes Doesn't Work

potatocubez, 2021-03-12:
I've accidentally updated my chrome os to version 89
I'm waiting for the next brunch update to fix my issue
Issues:
-Trackpad Button doesn't work most of the time
-Scrolling is inverted when swiping to the app drawer after the search/assistance tab or whatever its called.

Fixed issue:
Out of nowhere, the bluetooth issue where my phone won't show up in the bluetooth devices is fixed

SL-Pirate, 2021-03-12:
I think we all are experiencing that issue bro
Hope it will be fixed in the next brunch update
until then,use touchpad keys while holding touching the touchpad. Then the keys will work
Or else just use gestures aqs everyone does.

SL-Pirate, 2021-03-12:
BTW come join our [conversation](https://github.com/sebanc/brunch/issues/943)
we are talking about the same issue

potatocubez, 2021-03-12:
> BTW come join our [conversation](https://github.com/sebanc/brunch/issues/943)
> we are talking about the same issue

I did 🤣

SL-Pirate, 2021-03-12:
Damn U *****
I didn't notice 😂