sebanc/brunch #948: Back light issue

TheSwisscheese2013, 2021-03-12:
I have an asus t100ta with the latest ramsus and brunch installed. I have a screen brightness bar but it doesn't make adjustments when slid on the screen or with the keyboard keys. My main concern is that when I press my power button to turn off the screen, it goes black but the back light stays on, thus draining my battery within the hour.

b1t0y, 2021-03-13:
> I have an asus t100ta with the latest ramsus and brunch installed. I have a screen brightness bar but it doesn't make adjustments when slid on the screen or with the keyboard keys. My main concern [...]

Try using the kernel 5.10 or 4.19 by editing grub.cfg.    Coz in the default kernel backlight slider does not work but works when i tried 5.10 and 4.19.  Hope this helps.

TheSwisscheese2013, 2021-03-14:
> Try using the kernel 5.10 or 4.19 by editing grub.cfg. Coz in the default kernel backlight slider does not work but works when i tried 5.10 and 4.19. Hope this helps.

I apologize for my ignorance, it has been some time since I have modified kernels. But would you mind guiding me through the process. The search function has yielded me no results.

b1t0y, 2021-03-14:
> I apologize for my ignorance, it has been some time since I have modified kernels. But would you mind guiding me through the process. The search function has yielded me no results.

1. Press Ctrl + Alt + T to enter terminal
2. Type shell
3. Type and run sudo edit-grub-config.
4. Modify your grub entry by putting -4.19 or -5.10 after kernel... so it should look like "kernel-4.19" without the quotations.
5. Save by Ctrl+o (Press enter to confirm).
6. Exit by Ctrl+x.
7. Reboot.