sebanc/brunch #952: Surface Pro 4 issues

neo338388, 2021-03-14:
Laptop: Surface Pro 4 i5/4/128
Files used: brunch_r88_stable_20210223, chromeos_13505.111.0_rammus_recovery_stable-channel_mp-v2.bin

Issues:
1. Touch screen will not work after tablet sleep for a while, have to reboot to activate the touch screen again.
2. Power off is not a “real” power off.  Fully charged tablet after pressed power off, the next day will have 0% battery, can only power on when plugged in, and tablet shows battery charging from 1%.
3. Camera not working.  (Known issue)

I would like also to thank for your hardwork on this chromeOS project, it saved my lagging surface pro 4 when running window.

Appreciate your help especially on issues 1&2, fixing these 2 issues will help a lot.  Thanks.

vanth0me, 2021-06-20:
Yeah having the same issues. My Pro 3 is working fine.
I'm trying the new version now, maybe adding the new goodix(i dont know what it is) option, see if it helps.

Xeroka88, 2022-03-24:
> 

Hi, did you get fixes the camera?

ouija, 2022-04-03:
> Yeah having the same issues. My Pro 3 is working fine. I'm trying the new version now, maybe adding the new goodix(i dont know what it is) option, see if it helps.

Goodix is for fingerprint sensors as far as I know, not needed.   There is (limited) driver support at the kernel level for the cameras, and I don't think you'll be getting them to work anytime soon.

closed 2026-03-08 by github-actions[bot]