sebanc/brunch #958: how do i install chrome os?

kingmaj0r, 2021-03-16:
hi, can someone help me install chrome os on my pc?
right now I got windows 10.

pc specs:
GPU: Nvidia GeForce gtx 1060
CPU: ryzen 5
ram: 16GB
SSD: nvme 500GB
motherboard: msi

Andytheclown, 2021-03-16:
It’s not compatible with Nvidia. So that’s simple. Remove it and then it will probably install. Proably....

kingmaj0r, 2021-03-17:
rip will they soon support nidia

kingmaj0r, 2021-03-17:
> It’s not compatible with Nvidia. So that’s simple. Remove it and then it will probably install. Proably....

rip will they soon support nidia

kingmaj0r, 2021-03-21:
does it support it now? nvidia

potatocubez, 2021-03-21:
> does it support it now? nvidia

No.

kingmaj0r, 2021-03-21:
okay, thank you do you know when tho?

kingmaj0r, 2021-03-21:
> > does it support it now? nvidia
> 
> No.

okay, thank you do you know when tho?

zainullahk, 2021-05-31:
Never support will never come this is not official project

closed 2023-08-29 by kingmaj0r