sebanc/brunch #968: Trouble updating to the latest brunch release

JamesNoone69, 2021-03-22:
Hello!

How to know what release it currenting runing on your laptop ?

I have this message when I try to update to the last r89 unstable and I'm not sure the brunch version had changed.

[image]

Thanks!

WesBosch, 2021-03-26:
It looks like you updated to ChromeOS 90 at some point before updating to the latest brunch stable release. Unfortunately, ChromeOS 90 removes pv, which is used to updated brunch. (The new brunch release includes pv to fix this issue)

Currently, the simplest solution would be to backup what data you need to and reinstall using the latest brunch

J3n5en, 2021-04-05:
try to copy `pv` from `brunch_r89_stable_20210403.tar.gz/rootc.img/packages/binaries.tar.gz/usr/sbin/pv` to `/usr/bin` and re-run the update scripe. And it updated success for me.