sebanc/brunch #972: Google Play "something went wrong"

areeb4g, 2021-03-23:
I have finally got Chrome OS to work on my laptop (dual boot). Initially I had an issue where it would not even detect my touchpad. By adding i8042.nopnp-1 and pci=nocrs it all works including the stylus and touch screen. I honestly don't know what those commands do but I saw someone with an Acer Spin 5 say it works on their machine so I tried it. Now I have an issue where my Google Play store does not run. It gives the same error over and over again after many reboots and resets. It says "something went wrong" and it gives the option to try again or troubleshoot. Hitting try again, it says setting up but after about 10mins it returns to the "something went wrong screen" I have an Acer Spin 3 with i5 10th gen. I am currently on rammus with brunch 88. I have already tried Octopus and Coral to see if that would help.

areeb4g, 2021-03-23:
I saw on another issue page that someone asked to run the following command in crosh shell so hopefully this helps.
chronos@localhost / $ cat /var/log/messages | grep Android
2021-03-23T19:41:54.340723+00:00 INFO session_manager[1655]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2433
2021-03-23T19:47:11.987406+00:00 INFO session_manager[1655]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 2433
2021-03-23T19:47:39.773594+00:00 INFO session_manager[1655]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2433 stopped
2021-03-23T14:59:58.517180-05:00 INFO session_manager[1337]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2125
2021-03-23T15:02:08.141447-05:00 INFO session_manager[1337]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 2125
2021-03-23T15:02:34.786488-05:00 INFO session_manager[1337]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2125 stopped
2021-03-23T15:23:42.065659-05:00 INFO session_manager[1325]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2124
2021-03-23T15:24:47.001502-05:00 INFO session_manager[1325]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 2124
2021-03-23T15:25:13.559941-05:00 INFO session_manager[1325]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2124 stopped

princepjamie06832, 2021-03-24:
It seems Android container don't work in Ice Lake platform.

areeb4g, 2021-03-25:
> It seems Android container don't work in Ice Lake platform.
HP Chromebook x360 14c-ca0095nr. This chromebook seems to have an i5 10th gen processor. I'll try the hatch recovery image which the HP chromebook mentioned above uses. I hope this works.

sebanc, 2021-03-27:
arcvm has just been released for "hatch" recovery image and only on beta/dev channel. It seems to improve compatibility with previously unsupported 10th gen devices.

I will be interested to know if it works for you

princepjamie06832, 2021-03-27:
Wow, it works! Last time didn't worked Android container on Ice Lake platform. Right now is in beta and next month will rollout v90.

princepjamie06832, 2021-03-27:
#983

areeb4g, 2021-03-28:
Wow okay. I will close this thread cause that solved the issue.

closed 2021-03-28 by areeb4g