sebanc/brunch #973: Tablet mode not working on One Mix 1S (nocturne)

vladone97, 2021-03-24:
I'm using Chrome OS R89 on a One Mix 1S (Celeron 3965Y CPU). I built my Chrome OS image using the Pixel Slate recovery image. I've tried all methods(ctrl+alt+shift+t, sysfs_tablet mode, and force_tablet mode), but I can't switch to tablet mode. The only thing that happens is that the multitasking icon appears in the bottom right corner, but everything else stays the same.

kerta1n, 2021-03-26:
First up, you might want to downgrade to r88 stable. Secondly, you can choose from the following, but I've listed them in the order that they would work.

1. Octopus
2. Paine
3. Snappy
4. Reks
5. Setzer

Octopus is used by over 20 Chromebooks which (I think) use Celeron processors, but since you have a 3965, Paine might be better because those Chromebooks have a 4xxx CPU. I have an Inspiron 3542 with a Celeron N3050 CPU, so Setzer worked for me, but I will probably reinstall with Octopus as well, as Google is dropping support for updates for Setzer next year.

vladone97, 2021-03-26:
There is a version of the Pixel Slate with the exact same CPU as my One Mix, so compatibility shouldn't be an issue. I also tried the image from a Samsung chromebook which also had the exact same CPU (and on top of that, a 360 hinge just like the One Mix) on r83 a few months ago, and I had the same issue. I'll try these too, but in the meantime, I've discovered something else. On Windows, when I set it to automatically switch to tablet mode when a keyboard is not present, folding the display back doesn't switch the UI into tablet mode either. It might be a hardware thing.
I found out from someone on Telegram that Chrome OS can't force full tablet mode if it detects a keyboard, it only adds the multitasking button. It seems like folding the display back doesn't disconnect the keyboard, but rather just disables all keys except for the power button. If that's the case, would it be possible to add a way to switch to full tablet mode in Brunch?

kerta1n, 2021-03-30:
Sadly I don't think there is anything else in the available documentation that has focused specifically on tablet mode besides the options you listed, you could attempt "oled_display" and "alt_touchpad_config" but I'm not too knowledgeable about Brunch yet so I'll wait for someone else to come along and correct me on this.

closed 2026-03-05 by github-actions[bot]