sebanc/brunch #976: Infinite boot loop (Loading Brunch Framework)

Saroj-padhy, 2021-03-24:
Facing bootloop after installing chrome os in HDD.
I tried to install chrome os by using linux live image.
Installed it in a HDD completely.
Then it showed rootfs etc.
Then restarted and kept showing loading brunch framework and again restart. 
[image]
[image]

System specs:
Intel Pentium G series 4th gen (G3240)
No external GPU
4gb Ram

Rammus recovery 87 
Brunch , tried both 87 stable and 88 stable. 
Same error.

Saroj-padhy, 2021-03-25:
@sebanc help.

Saroj-padhy, 2021-03-25:
Ok it's solved when I changed kernel to 4.19. 😁

Saroj-padhy, 2021-03-25:
Restarting brings the problems again. How to edit grub permanently. ? @sebanc or any other #experts.

closed 2021-03-25 by Saroj-padhy

Roa1375, 2021-06-02:
> Ok it's solved when I changed kernel to 4.19. 😁

How do you change kernel version? I have same problem

MrZethaz, 2021-07-09:
> > Ok it's solved when I changed kernel to 4.19. 😁
> 
> How do you change kernel version? I have same problem

enter in debug mode  on boot and select kernel 4.19 :)

Roa1375, 2021-07-09:
> select

And how to enter debug mode?!

ASOwnerYT, 2021-07-16:
How do I enter debug mode???

AdityaBhati1, 2021-07-18:
in the grub menu press **E** and here you are in debug mode .
 
but i don't how to change kernel version ?

 would anyone help?

billal78, 2021-08-03:
> in the grub menu press **E** and here you are in debug mode .
> 
> but i don't how to change kernel version ?
> 
> would anyone help?

Hi, after pressing E (when you're in debug mode), press F2 then add -4.19 after kernel  (kernel-4.19) then boot by pressing F10
I hope i helped

Napoleon99999, 2021-10-02:
Please help what to write after grub>
I still can't understand how to change kernel.
Please tell step by step what to do after entering debugging mode. It shows grub>

cyberghost-whitedevil, 2021-12-28:
Press E and then use your keyboard and add -4.19 to kernel. Then press f10.. no need to press f2

mukulrajput112, 2022-01-31:
> Press E and then use your keyboard and add -4.19 to kernel present in the code. Then press f10.. no need to press f2

Bro this code is not saving in my pc and problem is not solved

Sherazus, 2023-07-20:
Can you chat with me in messenger.i have same problem.Plz reply.
My id link:https://www.facebook.com/sherazus.tahmeed

Sherazus, 2023-07-20:
> Facing bootloop after installing chrome os in HDD. I tried to install chrome os by using linux live image. Installed it in a HDD completely. Then it showed rootfs etc. Then restarted and kept [...]

> Ok it's solved when I changed kernel to 4.19. 😁

Can you chat with me in messenger. I have same problem.plz reply
My id: https://www.facebook.com/sherazus.tahmeed

Cybernautici74, 2025-01-01:
> Press E and then use your keyboard and add -4.19 to kernel present in the code. Then press f10.. no need to press f2

Io ancora no n riesco a capire come cambiare kernel. Potete aiutarmi con calma con i commandi o screenshot grazie

Cybernautici74, 2025-01-01:
> > in the grub menu press **E** and here you are in debug mode .
> > but i don't how to change kernel version ?
> > would anyone help?
> 
> Hi, after pressing E (when you're in debug mode), press F2 then add -4.19 after kernel (kernel-4.19) then boot by pressing F10 I hope i helped

Io non ancora riesco a capire come cambiare kernel. Mi potete dare i comandi con calma o uno screenshot grazie