sebanc/brunch #977: any idea of wich image use into an amd c-70?

mzramna, 2021-03-24:
ive installed cloudready into an old amd c-70,but i did not found wich of the recovery images work in there,since all the images are for intel or newer amd i dont know if it had any difference,i could use any or shoud try them ultil any work?do this may cause any problem to the hardware?

kerta1n, 2021-03-26:
The only AMD images out there for now are for the Ryzen and A4-A9 and E2. Their board names are Zork and Grunt, respectively. You could try those two, but if it doesn't work you could go with a lightweight distro like Lubuntu.

mzramna, 2021-03-26:
I'll try those,thanks,but the cloudready already Works in this computer,not 100% Fine ,but enough to use,so thanks but i think lubuntu ir other Linux Will not be necessary.
Btw: i've already used lubuntu 18.10 into this machine,the experience os Very similar to cloudready in relation to chromium

kerta1n, 2021-03-30:
> I'll try those,thanks,but the cloudready already Works in this computer,not 100% Fine ,but enough to use,so thanks but i think lubuntu ir other Linux Will not be necessary.
> Btw: i've already used lubuntu 18.10 into this machine,the experience is Very similar to cloudready in relation to chromium , a bit laggy when using more than 3 tabs , this is worse when are media sites,like Youtube,Netflix or Emby

Yeah, I faced this problem, so that's when I decided to install ChromeOS because there was already an image for my Celeron laptop.

mzramna, 2021-04-05:
i'll try to instal today this,any idea if using an wrong image could cause any hardware damage? or only will not work properly?

closed 2026-03-05 by github-actions[bot]