sebanc/brunch #979: Macbook Air 4,1 - no voice when playing video

memerich99, 2021-03-26:
Hi, I have a problem with the headphones on a Macbook Air 4,1. 
When playing video (Netflix, Amazon Prime, Disney) with headphones plugged in, I cannot hear any voices. Most likely it is a problem with downmixing, meaning that I do not get the center channel, whereas the left and right channels play fine.
This is however not the case on the internal speakers. On the internal speakers the downmixing seems to work and I can hear all sounds, also the voices.
Is there a setting in Chrome I can use to force downmixing also for the headphones? Or can I put kernel commands / options in grub.cfg to solve this issue?

kerta1n, 2021-03-26:
You could try installing an app from the webstore, but it may be due to you using the wrong image. From what I'm assuming, you are using a Core 2 Duo T8XXX series CPU. (if it isn't, please let me know in the reply.) What recovery image did you use? This also might have caused some issues as well.

memerich99, 2021-03-26:
The CPU in my Macbook Air is a Intel Core I7-2677M 1.8 GHz. The recovery image I used is Rammus r87, now updated to 89 via enable_update. The Brunch version is the latest beta: Brunch r89 unstable 20210325
But I have only updated the ChromeOS and Brunch framework because I experienced the problem also before updating. Unfortunately it has been persistent.
By the way, I am not sure the problem is related to the recovery image as I have also experienced it on Cloudready.

memerich99, 2021-03-27:
I have did some more testing. 
Whenever I switch the headphones in alsamixer to (M0) or (0M) then I can hear the voices from both speakers of the headphone. However the setting is not persistent, meaning that if I unplug the headphones they switch to (MM) and then when plugged back in to (00). But at setting (00) no voices do not come through.

closed 2026-03-05 by github-actions[bot]