sebanc/brunch #980: Android Container on HDD.

b4iterdev, 2021-03-26:
Well, just like other issue about android apps running on HDD, but this time it more worse. Now even I wait for container to be pre-loaded, it's now broken
```
2021-03-26T12:06:25.504861+00:00 INFO session_manager[1553]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2748
2021-03-26T12:09:08.715779+00:00 INFO session_manager[1553]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 2748
2021-03-26T12:28:01.651370+00:00 INFO session_manager[1553]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2748 stopped
2021-03-26T12:28:04.384223+00:00 ERR session_manager[1553]: [ERROR:session_manager_impl.cc(1354)] Error getting Android container pid.
2021-03-26T12:28:04.384315+00:00 ERR session_manager[1553]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerShutdownFail, Message=Error getting Android container pid.
2021-03-26T19:31:21.479665+07:00 INFO session_manager[1216]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2046
2021-03-26T19:31:21.508134+07:00 INFO session_manager[1216]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2046 stopped
2021-03-26T19:31:36.857888+07:00 INFO session_manager[1216]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2817
2021-03-26T19:32:24.713119+07:00 INFO session_manager[1216]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 2817
2021-03-26T20:35:43.429085+07:00 INFO session_manager[1216]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2817 stopped
2021-03-26T20:39:37.164523+07:00 INFO session_manager[1226]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 2663
2021-03-26T20:39:37.635075+07:00 INFO session_manager[1226]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 2663
2021-03-26T20:40:32.652505+07:00 INFO session_manager[1226]: [INFO:session_manager_impl.cc(1872)] Android container with PID 2663 stopped
2021-03-26T20:41:10.440256+07:00 ERR session_manager[1226]: [ERROR:android_oci_wrapper.cc(156)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2021-03-26T20:41:10.481331+07:00 ERR session_manager[1226]: [ERROR:session_manager_impl.cc(1779)] Starting Android container failed.
2021-03-26T20:41:10.481395+07:00 ERR session_manager[1226]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2021-03-26T20:43:33.973815+07:00 INFO session_manager[4537]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 5443
2021-03-26T20:43:37.092582+07:00 INFO session_manager[4537]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 5443
2021-03-26T20:44:11.330269+07:00 INFO session_manager[4537]: [INFO:session_manager_impl.cc(1872)] Android container with PID 5443 stopped
2021-03-26T20:48:31.640401+07:00 INFO session_manager[6357]: [INFO:session_manager_impl.cc(1785)] Started Android container with PID 6694
2021-03-26T20:49:28.452551+07:00 INFO session_manager[6357]: [INFO:session_manager_impl.cc(1280)] Android container is running with PID 6694
2021-03-26T20:49:49.669273+07:00 INFO session_manager[6357]: [INFO:session_manager_impl.cc(1872)] Android container with PID 6694 stopped
2021-03-26T20:50:17.703231+07:00 ERR session_manager[6357]: [ERROR:android_oci_wrapper.cc(156)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2021-03-26T20:50:17.744638+07:00 ERR session_manager[6357]: [ERROR:session_manager_impl.cc(1779)] Starting Android container failed.
2021-03-26T20:50:17.744698+07:00 ERR session_manager[6357]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
```
dmesg log: https://pastebin.com/r9cf3suh

aortiz294YT, 2021-04-07:
SSD or NVMe or eMMC is recommended hdd cant start that

khanhtranngoccva, 2021-08-30:
or wait for a minute.

closed 2026-03-05 by github-actions[bot]