sebanc/brunch #983: Android Container on 10th Gen Intel Compatibility

princepjamie06832, 2021-03-27:
Recently updated to v90 in beta, it seems Android Container is working okay. It's called ARCVM and running Android 11.

**Before v90,** when I attempting to open Play Store. But I got error encountered, it said 'Something went wrong'. It seems Android Container don't work in Ice Lake Platform.
**In v90,** my second attempt to open Play Store and it working. After updated, error no longer present when opening Play Store.

If you interested to test v90, switch to Beta channel and only available for 'Hatch'. 
Next month will rollout v90 stable and comes with Android 11. Remember, don't pick wrong recovery image, otherwise Play Store may not open without ARCVM. Any encounter error? Let us know.

rvaldomir, 2021-04-25:
i have i51021u gen and decided to try puff instead rammus, as with rammus i was experiencing a lot of hangs / freeze and serveral unexpected restarts.

maybe you can try with that one. 

it is more for chromebox, i decided to use that as i have NUC box.

but i think for i5 10th gen in laptops you can use hatch as the ASUS Chromebook Flip C436FA uses 10th gen of intel

princepjamie06832, 2021-04-25:
> i have i51021u gen and decided to try puff instead rammus, as with rammus i was experiencing a lot of hangs / freeze and serveral unexpected restarts.
>
> maybe you can try with that one.
>
> it is more [...]

If you have desktop version with 10th Gen Intel, you can try 'Puff' recovery image but v90 slowly to release for 10th gen.

princepjamie06832, 2021-05-16:
I'm closing this issue. It seems Android Container on 10th Gen Intel is working perfectly. Since I tried out unstable Kernel 5.4, 
Google Play Store was not working until now.

closed 2021-05-16 by princepjamie06832