sebanc/brunch #988: trouble with touchscreen on chuwi corebook

bplane, 2021-04-01:
Hello.

I have a chuwi corebook. Almost everything works as far as I noticed. But the touchscreen does not correctly handle touching, when I do a long touch or do not remove my finger from the screen, it processes it not as one long press, but as a lot of short ones. how can i fix it?

forumi0721, 2021-04-22:
You need to check your tablet touchscreen is silead.
(I know about silead only...)
"dmesg | grep silead" shows firmware error, you need to add firmware.

Refer to gsl_firmware, download or extract firmware.

https://github.com/onitake/gsl-firmware

darkIncorporative, 2021-07-27:
@forumi0721 
I am a noob at this. I have the fw and firmware files i just dont know how to install it on chrome or do i need to create a new image or something