sebanc/brunch #989: RTL8723BE support

KazuharuAI, 2021-04-02:
Could you add support to rtl8723be?
Hello, I recently installed the system on a laptop but the network card does not work, then I tried to enable with the most recent version r89 unstable 20210327, even try with the most similar ones rtl8723du, rtl8723bu and rtl8723de, but apparently it remains the same.

sebanc, 2021-04-07:
The rtl8723be is directly included in the mainline kernel (no option needed), are you sure this is your wireless adapter model ?

KarelAI, 2021-04-09:
Thank you for responding and taking advantage of the opportunity I would like to apologize for something minuscule and that I did not realize before making the post. After doing some research on the adapter model, I discovered that the problem was where the antenna was installed, apparently it was at 2 and not 1, I just moved it around and it worked.

Again, I'm sorry for wasting your time.

sebanc, 2021-04-09:
No pb, happy to see that it works now :)

closed 2021-04-09 by KarelAI