sebanc/brunch #990: Experimenting with your unstable releases

Lucaacer, 2021-04-03:
As you might remember, I was was using your previous unstable with the option no camera config, which worked very well.

Just to try it, I flashed your latest unstable which should improve wifi stability and this thing looks stable.

On the other side, bluetooth does not work, it's grayed out in settings and it's not being detected at system level.

Here are some logs:
BluetoothEvent[07:19:34] Releasing Adapter.

BluetoothEvent[07:19:34] Event Listener Removed: os-settings Count: 0

BluetoothEvent[07:19:12] Event Listener Added: os-settings Count: 1

BluetoothUser[07:19:12] BluetoothEventRouter()

BluetoothEvent[07:19:12] BluetoothAPI: Creating BluetoothEventRouter

BluetoothEvent[00:44:52] Applying primary user pref bluetooth power: 1

BluetoothEvent[00:44:52] BluetoothAPI: 0x24c2fb2a3800

BluetoothEvent[02:44:10] Pairing agent now default

BluetoothEvent[02:44:10] Pairing agent registered, requesting to be made default

BluetoothEvent[02:44:08] Bluetooth adapter ready, IsPresent = 0

BluetoothEvent[02:44:08] BlueZ Adapter Initialized.

BluetoothEvent[02:44:08] BluetoothAPI: 0x24c2f965a000

BluetoothEvent[02:44:08] Applying local state pref bluetooth power: 1

and this one
chronos@localhost / $ rfkill
bash: rfkill: command not found
chronos@localhost / $ sudo rfkill
ID TYPE      DEVICE                   SOFT      HARD
 0 wlan      phy0                unblocked unblocked
 2 bluetooth tpacpi_bluetooth_sw unblocked unblocked
 3 bluetooth hci0                unblocked unblocked
chronos@localhost / $ hcitool dev
Devices:
chronos@localhost / $ sudo hcitool dev
Devices:
chronos@localhost / $ hcitool -i hci0 scan
Invalid device: Network is down

Just to report, before it goes to stable.

Have a good Easter.

sebanc, 2021-04-03:
Indeed I noticed it too, could you try the new brunch-unstable release ?

Lucaacer, 2021-04-03:
Yes, I'm downloading it!

Lucaacer, 2021-04-03:
Bluetooth works, thank you.

By the way, I am on the beta channel build  90.0.4430.51.

closed 2021-04-07 by Lucaacer