sebanc/brunch #996: Displaylink doesn't work on latest stable

daemongloom, 2021-04-07:
HP Spectre x2.

Displaylink device is not activated on latest stable brunch. Works fine with brunch_r88_stable_20210223.

Dmesg is attached.
[dmesg_r88_20210407_0815.log](https://github.com/sebanc/brunch/files/6268749/dmesg_r88_20210407_0815.log)
[dmesg_r89_20210407_0812.log](https://github.com/sebanc/brunch/files/6268750/dmesg_r89_20210407_0812.log)

daemongloom, 2021-04-08:
I've installed new unstable build.
Now it crashes after display connection and reboots system. I wasn't able to get log as "dmesg -w" gave me empty file after crash.

daemongloom, 2021-04-09:
The same happens with r89 unstable 20210409.

daemongloom, 2021-04-12:
The same happens with r89 unstable 20210410.

daemongloom, 2021-04-16:
The same happens with "brunch_r89_20210416.tar.gz"

daemongloom, 2021-05-11:
Brunch r90 unstable 20210509
Chrome OS 90.0.4430.100

There is no crash anymore, but external display is not recognized. 
Looks like https://github.com/sebanc/brunch/issues/1046 is a duplicate of this issue.

sebanc, 2021-05-13:
Could you post a dmesg after plugging the displaylink adapter ?

daemongloom, 2021-05-13:
I'm sorry, can't test it without using adapter completely (as it is my power source). I can give dmesg after plugging HDMI to that adapter, but the only message it give is 
"evdi: [E] add_store:257 Invalid device count "usb:2-4.2""
Twice on first HDMI connection, once on second HDMI connection. Unplugging HDMI gives nothing.
[dmesg_r90_20210513_1103.log](https://github.com/sebanc/brunch/files/6470617/dmesg_r90_20210513_1103.log)
[dmesg_r90_20210513_1058.log](https://github.com/sebanc/brunch/files/6470619/dmesg_r90_20210513_1058.log)

I'll try to find another power adapter and post a new dmesg.

daemongloom, 2021-05-13:
The same happens with new version of brunch.

sebanc, 2021-05-14:
Could you try the new brunch-unstable build ?

daemongloom, 2021-05-14:
I'll check it on Monday. If it won't work - I'll be able to post dmesg without displaylink attached.

daemongloom, 2021-05-17:
Brunch r90 unstable 20210516

It works, but dmesg shows following crash each time cursor is moved between screens:

```
[  179.238845] ------------[ cut here ]------------
[  179.238868] WARNING: CPU: 0 PID: 6657 at drivers/gpu/drm/drm_gem.c:1035 drm_gem_object_put+0x1f/0x32 [drm]
[  179.238869] Modules linked in: vhost_vsock vmw_vsock_virtio_transport_common vsock vhost evdi hp_wmi intel_xhci_usb_role_switch wmi xt_cgroup intel_lpss_acpi tpm_tis tpm_tis_core intel_hid intel_vbtn acpi_pad iio_trig_sysfs hid_sensor_cros_compat industrialio_triggered_buffer kfifo_buf hid_sensor_iio_common industrialio hid_sensor_hub cros_ec_ishtp intel_ishtp_hid snd_usb_audio snd_usbmidi_lib snd_rawmidi cdc_mbim cdc_wdm cdc_ncm btusb btrtl btbcm btintel ov5670 snd_hda_codec_hdmi snd_hda_codec_conexant mei_hdcp snd_hda_codec_generic ledtrig_audio snd_soc_skl snd_soc_sst_dsp snd_soc_sst_ipc snd_hda_ext_core snd_hda_intel snd_intel_nhlt mei_me snd_hda_codec iwlmvm snd_hda_core mei mac80211 snd_hwdep ipu3_imgu(C) iwlwifi intel_lpss_pci intel_lpss ipu3_cio2 v4l2_fwnode idma64 intel_pch_thermal videobuf2_dma_sg intel_ish_ipc cfg80211 intel_ishtp cdc_ether usbnet r8152 mii i915 video intel_gtt i2c_algo_bit drm_kms_helper cec drm
[  179.238904] CPU: 0 PID: 6657 Comm: DrmThread Tainted: G        WC        5.4.109-brunch-sebanc #1
[  179.238905] Hardware name: HP HP Spectre x2 Detachable        /8146, BIOS F.34 12/04/2019
[  179.238927] RIP: 0010:drm_gem_object_put+0x1f/0x32 [drm]
[  179.238929] Code: 5b 5d 41 5c 41 5d 41 5e 41 5f c3 0f 1f 44 00 00 48 85 ff 74 27 55 48 8b 47 08 48 89 fd 48 8d 78 68 e8 a0 39 6b dd 84 c0 75 02 <0f> 0b 48 89 ef 48 c7 c6 25 44 29 c0 5d e9 74 fb ff ff c3 48 85 ff
[  179.238931] RSP: 0018:ffff971041e57bd8 EFLAGS: 00010246
[  179.238933] RAX: 0000000000000000 RBX: ffff899f555c6980 RCX: 0000000000000000
[  179.238935] RDX: ffff899f7489bf00 RSI: ffff971041e57bc8 RDI: ffff899f57cda068
[  179.238936] RBP: ffff899f68785c00 R08: 0000000000000001 R09: 0000000000000000
[  179.238938] R10: 0000000000000000 R11: 0000000000000020 R12: 0000000000000000
[  179.238939] R13: 0000000000000000 R14: ffff899f555c69b0 R15: 0000000034325241
[  179.238941] FS:  00007cee96536700(0000) GS:ffff899faaa00000(0000) knlGS:0000000000000000
[  179.238960] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  179.238961] CR2: 000020f04348e200 CR3: 00000001347b2004 CR4: 00000000003606f0
[  179.238963] Call Trace:
[  179.238973]  evdi_cursor_set+0xc1/0xdb [evdi]
[  179.238980]  evdi_crtc_cursor_set+0xca/0x102 [evdi]
[  179.239023]  drm_mode_cursor_common+0x14a/0x1e5 [drm]
[  179.239049]  ? drm_mode_setplane+0x216/0x216 [drm]
[  179.239073]  drm_mode_cursor_ioctl+0x3f/0x59 [drm]
[  179.239110]  drm_ioctl_kernel+0x92/0xe6 [drm]
[  179.239131]  drm_ioctl+0x249/0x337 [drm]
[  179.239152]  ? drm_mode_setplane+0x216/0x216 [drm]
[  179.239156]  ? ___might_sleep+0x47/0x146
[  179.239161]  vfs_ioctl+0x1f/0x2d
[  179.239164]  do_vfs_ioctl+0x6be/0x6f0
[  179.239167]  ksys_ioctl+0x4e/0x6d
[  179.239170]  __x64_sys_ioctl+0x16/0x19
[  179.239174]  do_syscall_64+0x4b/0x59
[  179.239177]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  179.239180] RIP: 0033:0x7cee9c2847c7
[  179.239183] Code: c3 66 90 48 8b 05 d1 86 0b 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a1 86 0b 00 f7 d8 64 89 01 48
[  179.239184] RSP: 002b:00007cee96534c28 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  179.239187] RAX: ffffffffffffffda RBX: 00007cee96534ea0 RCX: 00007cee9c2847c7
[  179.239188] RDX: 00007cee96534c68 RSI: 00000000c01c64a3 RDI: 000000000000002a
[  179.239189] RBP: 00007cee96534c50 R08: 0000000000000000 R09: 0000000000000000
[  179.239190] R10: 0000000000000002 R11: 0000000000000246 R12: 000000000000002a
[  179.239192] R13: aaaaaaaaaaaaaaaa R14: 00007cee96534c68 R15: 00000000c01c64a3
[  179.239195] ---[ end trace c6cb8646d5a9bbcb ]---
```

And it looks like I'm missing crostini now. Going to reboot and/or restore to previous versions to check.

sebanc, 2021-05-17:
That's very good news. I think those warnings can safely be ignored if they do not create specific issues.

Regarding crostini, I don't think something in the new release could cause that but it has recently been moved to the advanced settings tab.

You use the default 5.4 kernel right ?

daemongloom, 2021-05-17:
It works fine now, performance is the same as in previous release.

Crostinig was missing because of issues with my hardware, sorry. Hardware Virtualization flag in bios got reset.

daemongloom, 2021-05-31:
Works fine with latest stable version. Closing this issue.

closed 2021-05-31 by daemongloom