sebanc/brunch #999: Stuck to ChromeOS - Screen

Konarer, 2021-04-09:
Hi all, I am new to all of this (linux, ChomeOS, github, ... all .....), so sorry if I am doing something wrong in this channel.

I am able to install everything out of linux as it should be and i get within Linux (Mint) that ChromeOS is installed. When rebooting the system, Brunch starts does some things and then ChromeOS starts. But then I get stuck at this screen:
[image]

I am trying to isntall ChromeOS for my family on a HP EliteBook 8770w.
Specs:
Intel Core-i5 3360M
8GB RAM
Chip: Mobile Intel® QM77 Express

I tried several images (ramus, samus, octopus, coral, ....) and different release of brunch (86, 87, 88, 89) with the coresponding images. With all of em I get stuck to the ChromeOS screen. 

Any suggestions or ideas what I can do to fix this?
Many thanks in advanced.

Regards 
Joe

Sayor0630, 2021-04-09:
Which generation of Intel CPU is this?

Sayor0630, 2021-04-09:
If your CPU generation is equal to or lower than the 3rd generation Intel i5 CPU, then you should and must use SAMUS recovery image. Otherwise, it won't work properly.

Sayor0630, 2021-04-09:
Use this image. Download by clicking this link.
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_13729.56.0_samus_recovery_stable-channel_mp-v3.bin.zip

Konarer, 2021-04-09:
Yes, it is 3. Genernation. I used Samus first and it didnt work so i tried all the rest with the same result.

Sayor0630, 2021-04-09:
You can try changing brach karnel version to karnel 4.19

Konarer, 2021-04-09:
I did this within the grub.cfg and I am not sure if I made it correctly...
[image]

I dont see a change in while booting.
Can I see during brunch-logo, which kernel is used?

Sayor0630, 2021-04-09:
By defaut, branch use Karnel 5.4

Sayor0630, 2021-04-09:
> I did this within the grub.cfg and I am not sure if I made it correctly...
> [image]
> 
> I dont see a change in while booting.
> Can I see during brunch-logo, which kernel is used?

Yes, this is correct.

Sayor0630, 2021-04-09:
But, again I am saying
**_replace "/kernel" on the grub configuration linux line with "/kernel-4.19_**

Konarer, 2021-04-09:
I did, the ebug showed also that it started with kernel-4.19 but it didnt help. Still suck on Chrome Screen

sebanc, 2021-04-09:
I guess your laptop has a dual gpu, if so the best solution would be to disable the dual gpu in the bios but you can also try adding "module_blacklist=nouveau" on the kernel command line.

Sayor0630, 2021-04-09:
Do you have dual GPU?

Sayor0630, 2021-04-10:
> I guess your laptop has a dual gpu, if so the best solution would be to disable the dual gpu in the bios but you can also try adding "module_blacklist=nouveau" on the kernel command line.

You can use this command. See if it works.

Sayor0630, 2021-04-10:
- AMD FirePro M4000 with 1GB dedicated GDDR5 video memory
 Microsoft DirectX 11 (Shader Model 5.0) and OpenGL 4.1/4.1+ capable

- AMD Eyefinity Technology supports up to five independent displays 
when using an HP Advanced Docking station

- NVIDIA Quadro K3000M with 2GB dedicated GDDR5 video memory
Microsoft DirectX 11 (Shader Model 5.0) and OpenGL 4.2 capable

- NVIDIA Quadro K4000M with 4GB dedicated GDDR5 video memory
Microsoft DirectX 11 (Shader Model 5.0) and OpenGL 4.2 capable

- NVIDIA Quadro K5000M with 4GB dedicated GDDR5 video memory   
Microsoft DirectX 11 (Shader Model 5.0) and OpenGL 4.2 capable

Do you have all of these?

Konarer, 2021-04-10:
Hi guys, thanks for all your help. But looks like, i killed the patient.

I tried to update BIOS to see, if I would be able to disable the videocard in a newer bios version. The first boot was without problems and the bios update too, but since today i am not able to boot the computer anymore.

I tried fb- commands and module_blacklist=amdgpu and it didnt help.

The command "module_blacklist=nouveau" is new to me, but since the computer is broken now, I won't be able to test this :/. 
All hp bios recovery options tried and none did help. Screen stays black. 

Sad but no problem at all. I will find a new patient soon ;).
Thanks for all help.

closed 2021-04-10 by Konarer

Sayor0630, 2021-04-12:
You should again flash your BIOS.

Sayor0630, 2021-04-12:
Download your BIOS from your laptop manufacturer's website. Then use a Pendrive to flash it.

Sayor0630, 2021-04-12:
If it doesn't work, then format the drive of ChromeOS using windows or Linux bootable pendrive