sebanc/brunch #1002: rtl8723be wifi and bluetooth not working

abhiatar, 2021-04-11:
you are great my old laptop become extremely fast with chrome os
but my wifi and bluetooth not workin
my laptop model is hp 15q-bu009tu
and i think my card model is rtl8723be
please help me sebanc

i will always be grateful to you

abhiatar, 2021-04-11:
Please elaborate I know very less about thi kernel

MelliCarvalho, 2021-04-12:
I'm also having trouble with bluetooth. I use a wireless mouse and keyboard. Sometimes the mouse stops responding and only works again when rebooting the computer. Sometimes when I turn off and turn on bluetooth, he responds again.

aishivam, 2021-04-20:
Try this : 

git clone https://github.com/lwfinger/rtlwifi_new.git

cd rtlwifi_new/ && git checkout origin/extended -b extended

sudo apt-get install linux-headers-generic build-essential git

sudo make install

sudo modprobe -r rtl8723de
sudo modprobe rtl8723de

closed 2026-03-04 by github-actions[bot]