sebanc/brunch #1004: Ram Always at 7.4 gb/8 gb

mittal002, 2021-04-13:
@sebanc for a few days chromeos was woking fine. but then now chromeos takes 7.4 gb ram out of 8 all the time. Its versy slow now. Even after reset reinsteall etc. It is installed on laptop not on pendrive.

daemonspudguy, 2021-04-16:
1. RAM usage graphs aren't particularly useful on Linux
2. Chrome uses a lot of computer resources.
3. You might want to enable performance-related flags in chrome://flags.

asukumar10248, 2021-05-22:
Those are not options, just add them on the kernel command line after "cros_debug" and before "options=...." if any:

"enforce_hyperthreading=1": improve performance by disabling a ChromeOS security feature and forcing hyperthreading everywhere
this will help you