sebanc/brunch #1008: Computer freezes for no reason

kissunyeason, 2021-04-16:
My computer freezes from time to time. At first, there is no response after clicking the mouse, but the mouse can move. After a few seconds, the entire system becomes unresponsive and can only be forced to shut down.

CHROME VERSION |   | 91.0.4468.3 dev

CHROMEOS_RELEASE_DESCRIPTION |   | 13900.0.0 (Official Build) dev-channel zork

brunch R89

CPU ryzen 2500U

G-Ventur4, 2021-04-20:
The same thing is happening with me

rvaldomir, 2021-04-22:
Same happening here but with chrome. 90 stable channel brunch r89

Does anyone knows why it happens?

rvaldomir, 2021-04-25:
I think i fixed it by moving to another image 

as i have i51021u i decided to try puff instead rammus, as with rammus i was experiencing a lot of hangs / freeze and serveral unexpected restarts.

i did a small research check a chromebox (in my case as i have a NUC). that i most similar to the HW i have

and it is working better.

kissunyeason, 2021-04-26:
> I think i fixed it by moving to another image
>
> as i have i51021u i decided to try puff instead rammus, as with rammus i was experiencing a lot of hangs / freeze and serveral unexpected restarts.
>
> i [...]

my cup is ryzen 2500U，which image i can use except zork

asukumar10248, 2021-05-22:
Those are not options, just add them on the kernel command line after "cros_debug" and before "options=...." if any:

"enforce_hyperthreading=1": improve performance by disabling a ChromeOS security feature and forcing hyperthreading everywhere
this will help you

kissunyeason, 2021-05-26:
> Those are not options, just add them on the kernel command line after "cros_debug" and before "options=...." if any:
> 
> "enforce_hyperthreading=1": improve performance by disabling a ChromeOS security feature and forcing hyperthreading everywhere
> this will help you

I have already added it, but my computer's fan keeps running. Is this caused?

asukumar10248, 2021-05-26:
add this between cros_debug and options
module_blacklist=nouveau

IsaacFHL, 2021-06-21:
Hello everyone! Finally i fixed this issue. if you have a Ryzen and a NVMe or SSD you need add this on grub:

iommu=soft

Thanks a a distro of Linux i can see error of this issue:

nvme amd-vi event logged io_page_fault

i found this issue in Internet and i fixed with this parameter. I see a kernel bug of Linux.

https://tutorials.technology/solved_errors/AMD_Vi_IO_PAGE_FAULT_Fix.html

kissunyeason, 2021-07-30:
> Hello everyone! Finally i fixed this issue. if you have a Ryzen and a NVMe or SSD you need add this on grub:
>
> iommu=soft
>
> Thanks a a distro of Linux i can see error of this issue:
>
> nvme amd-vi event [...]

This is of no use, and my computer crashes even more frequently.

kissunyeason, 2021-07-30:
> add this between cros_debug and options
> module_blacklist=nouveau

I don't have an additional graphics card