sebanc/brunch #1011: Please i want speed reply

3ala2khaddour, 2021-04-16:
Please i want to ask can i install chrome os 89 on legacy bios by dual boot with windows 7 on separate local drive example : my windows 7 was installed on drive C and want to install chrome os on local drive D and both of this local drives C & D at the same Hardisk / if it possible please tell me how and thanks so much
@@@@@@@@@@@
I have problem i tried to install chrome os on acer travelmate 5330 legacy bios i have 3 local drive C and D and H on drive C there is windows 7, and i formated drive d and delete it to be unlocatedSpace then in Linux mint i use gparted and make it primary and ntfs in Linux d drive name sda3 55 gigabytes i tried to install chrome os on it so when i reach to finall step i write on terminal      Sudo bash chromeos-install.sh - srcs samus.bin - dst /dev/sda3  put the terminal don't accept to install chrome os on local drive D (sda3).
@@@@@@@
Information my laptop that i tried to download chrome os on it is acer travelmate 5330 legacy bios have intel celeron T3100 1,9 GHz dual core L2 1 mb 64bit DDR2  (2009) 
&& Graphic prosseror Intel GMA 4500M (generation 4 // 2008)

ameyjoshi3068, 2021-04-17:
If you have legacy bios, you cannot dual boot. you will have to install Chrome OS in whole mechine. This will deleat windows.
Extract the mbr_support.tar.gz (overwriting files when requested) in the same folder
and paste following lines.
`sudo bash chromeos-install.sh -src samus.bin -dst /dev/sda`

NOTE: This command is according to the information you provided above. If this doesn't work, you will have to provide me a screenshot of folder which have all files and output of `sudo lsblk` in terminal.

ameyjoshi3068, 2021-04-17:
Just reply when it is done.

closed 2026-03-04 by github-actions[bot]