sebanc/brunch #1012: Touch screen not working

neo338388, 2021-04-17:
Added ipts options, touchscreen worked well on startup, but once idle for a while, touchscreen will not work at all, have to reboot, but idling a while is something cannot be avoided...

brunch and OS 88 version
SP4

LegacyGamerHD, 2021-04-27:
i also noticed this yes

LegacyGamerHD, 2021-04-27:
> Added ipts options, touchscreen worked well on startup, but once idle for a while, touchscreen will not work at all, have to reboot, but idling a while is something cannot be avoided...
> 
> brunch and OS 88 version
> SP4

i would set so it only turns of the screen

carbog, 2021-05-10:
Have you tried Windows + Shift + t to enable the touchscreen?

neo338388, 2021-05-12:
> Have you tried Windows + Shift + t to enable the touchscreen?

Tried but still not working, only solution is reboot.