sebanc/brunch #1013: Wifi doesnt work after sleep

hehecoi222, 2021-04-17:
After updated to r89 image and brunch, my Surface Pro 3 cant connect to the internet after sleep. When running Connectivity Diagnostics it showed that Captive Portal was in "warning" status. I could connect to every wifi spots but all of them couldnt connect to the internet, while ethernet is still working good.
Shutdown could solve this problem but it is not convenient to do that all the time

asukumar10248, 2021-04-24:
it is embeded in chrome os to turn of wifi in sleep mode

ecaron, 2021-04-24:
@hehecoi222, would you feel comfortable closing this as a duplicate of #953? (Which itself is a duplicate of #42, and the recent work on https://github.com/linux-surface/kernel/issues/20 might squash this once and for all...)

hehecoi222, 2021-04-25:
ive followed the solution of # 42 but the problem was not fixed

ecaron, 2021-04-25:
@hehecoi222 I understand that 42 didn't fix it. My recommendation was closing this as a duplicate of 953, so more eyes are on that ticket (which is where I'm writing my progress reports).

closed 2021-04-27 by hehecoi222