sebanc/brunch #1014: Disk Partition Mount

BilinmeyenKomut, 2021-04-18:
Hi

I'm install brunch my SD card. My SD card 256 GB. I'm use 200 GB for my personal files. But i cant mount this partition (sdb13)

How can i mount this partition? I'm try mount /media/removable but now showing in file manager. I'm use mount_internal_drive options alreday. Sory my bad english. Thanks.
[image]

ameyjoshi3068, 2021-04-18:
I have mentioned it here. By this, you don't even need to use mount_internal_drives
https://github.com/sebanc/brunch/issues/409#issuecomment-723944962

BilinmeyenKomut, 2021-04-18:
> I have mentioned it here. By this, you don't even need to use mount_internal_drives
> [#409 (comment)](https://github.com/sebanc/brunch/issues/409#issuecomment-723944962)

Thanks it's working. how can i automatic this for reboot or shutdown?

ameyjoshi3068, 2021-04-19:
Just search it on google. You will find tons of articles for this.