sebanc/brunch #1016: touch screen is flipped(or reverse)

chochoi, 2021-04-20:
hi, i use lg 2 in 1 tablet pc 10t370
(with touch screen),(bios and grub is rotate 90°)
and i finally install chrome os
but i cant use touchscreen because it reverse
for example i touch up-right
in chrome it is down-left
i dont know how can i fix it help

and first boot chrome, chrome logo is reverse(it rotate 270°)
and turn chrome os is fine(90° rotate)
in chrome i can rotate well but can not use touch

forumi0721, 2021-04-22:
Hi, I found a method for touchscreen property.
But I don't know about this is working.
TOUCHSCREEN_PROPRTIES built in kernel so I think add kernel parameter works propertly.

Refer to
https://www.kernel.org/doc/Documentation/devicetree/bindings/input/touchscreen/touchscreen.yaml

chochoi, 2021-04-22:
> Hi, I found a method for touchscreen property.
> But I don't know about this is working.
> TOUCHSCREEN_PROPRTIES built in kernel so I think add kernel parameter works propertly.
> 
> Refer to
> https://www.kernel.org/doc/Documentation/devicetree/bindings/input/touchscreen/touchscreen.yaml

thanks for reply, but I don't know how to apply you method, what to search for this? and i should use linux?

chochoi, 2021-04-23:
ctrl alt t and add shell and sudo edit-grub-config

so i add at back options=,  touchscreen-inverted-x and y but it doesnt work for me.
im right? or i mistake for some thing?

nlibby987, 2021-10-16:
I am having the same issue, any idea if this fixed the issue and what was the proper formatting to enter in the feilds

nlibby987, 2021-10-16:
Attached file as requested, when booting the unit is 90 degrees clockwise and I have rotated it in the chromeos settings but that also rotates the touch screen issue with it (no matter what orientation it is fully inverted).

 I also attached hardware info from brunch-toolkit if it is helpful. 

Thanks

[dmesg.txt](https://github.com/sebanc/brunch/files/7357754/dmesg.txt
[brunch-hwcomp.log.txt](https://github.com/sebanc/brunch/files/7357755/brunch-hwcomp.log.txt)
)

nlibby987, 2021-10-17:
> @nlibby987<https://github.com/nlibby987> Could you try the latest brunch-unstable release:
> https://github.com/sebanc/brunch-unstable/releases/tag/r94-unstable-20211017

Done, tried with just updating brunch, then tried it with both Chrome OS and Brunch updated and the touchscreen is still inverted. One thing did change the orientation of the display changed I know have to rotate the screen to 180° for it to be correct instead of 90° like it was before the update.

sebanc, 2021-10-17:
Is the touchscreen still inverted the same way as before ?

nlibby987, 2021-10-17:
Yes, if I touch the top right corner it responds at the bottom left corner.

sebanc, 2021-10-20:
Sorry for the late reply.

Could you try to add boot with "of_touchscreen.custom=1 of_touchscreen.invert_x=1 of_touchscreen.invert_y=1" added on the kernel command line ?

nlibby987, 2021-10-20:
Tried and it did not seem to make any change, entered the changes via sudo edit-brunch-config command and rebooted.

[image]

CitizenXVIL, 2022-02-02:
If it helps, the same occurs with my EVOO tablet. If I touch the bottom left of the screen, it registers as the top right. This is in Chrome. 
I tried the kernel commands but they unfortunately didn't fix it either. If I can provide any information to help fix this, I'd be happy to try.

amon-666, 2022-03-18:
same occurs with my galaxy book 10.6 tablet

Naced28, 2022-07-02:
Hi, I have the same problem on my Chuwi hi10 go. I installed ubuntu 22.04 to see if I have the same problem and yes. The touchscreen is inverted X & Y.
I install the touchscreen calibrate tool from the ubuntu store to generate a 99-calibration.conf

In chrome OS, how can I send it to the etc/gesture ? Should I have to rename it "99-touchscreen.conf" ?
Thanks in advance for your answer

i-cry, 2025-08-11:
> hi, i use lg 2 in 1 tablet pc 10t370 (with touch screen),(bios and grub is rotate 90°) and i finally install chrome os but i cant use touchscreen because it reverse for example i touch up-right in chrome it is down-left i dont know how can i fix it help
> 
> and first boot chrome, chrome logo is reverse(it rotate 270°) and turn chrome os is fine(90° rotate) in chrome i can rotate well but can not use touch brunch 89, chrome os samus 89 (both lastest)

Were you able to fix it? I'm having the same problem