sebanc/brunch #1024: Unable to turn on Wi-Fi

Nakul-TK, 2021-04-26:
My Wi-Fi is Broadcom Corporation BCM43142 802.11b/g/n (rev 01)
I have installed Chrome OS in my PC single boot
My Wi-Fi is not working

[image]

If i turn on the Wi-Fi  it is showinf "Wi-Fi is turned off"
How to resolve this problem ?

Nakul-TK, 2021-04-26:
Please Reply

Nakul-TK, 2021-04-26:
Please Reply

truongtam-fudosan, 2021-04-26:
https://github.com/sebanc/brunch#framework-options

truongtam-fudosan, 2021-04-26:
@Nakul-TK 
Use Crtl + Alt + T -> Input `shell` ->Input `sudo edit-grub-config`

Nakul-TK, 2021-04-26:
It is asking password after sudo edit-grub-config

Nakul-TK, 2021-04-26:
please reply

Nakul-TK, 2021-04-26:
what is the password of chronos@localhost

Nakul-TK, 2021-04-26:
It is asking password after sudo edit-grub-config
what is the password of ***@***.***

Nakul-TK, 2021-04-26:
reply please

Nakul-TK, 2021-04-26:
[image]

What is the password ?

Nakul-TK, 2021-04-26:
I have tried , but please can you send image while adding "options=broadcom_wl"

ecaron, 2021-04-26:
@Nakul-TK, please close #1026 - duplicate tickets just slow down the speed that you get support. Also I'd suggest reading https://github.com/kossnocorp/etiquette - support on this channel is free, 9 posts in 8 hours is a lot of noise. People tend to prefer one really well written comment versus 8 one-liners.

If you're being asked for a root password on your machine but you don't remember setting it, then you should probably Powerwash your machine and try again (CTRL+ALT+SHIFT+R at the login screen). **Sidenote: Powerwashing, as you may know, will wipe all data from your machine & put it back in factory reset state as if you just installed Brunch for the first time**

Once you can successfully open crosh, then shell, then sudo, the instructions at https://github.com/sebanc/brunch#from-chromeos talk you through what you need.

Nakul-TK, 2021-04-27:
I havd done that but WiFi is not working.
We can do anything with dmesg.txt ?

Nakul-TK, 2021-04-27:
[dmesg.txt](https://github.com/sebanc/brunch/files/6381617/dmesg.txt)
I got this as output

Please help me

Nakul-TK, 2021-04-27:
please please reply
its urgent

seyoon20087, 2021-05-16:
> what is the password of chronos@localhost

Try using the password "chrome"

Nakul-TK, 2021-05-16:
i powerwashed and it didnt ask password . 
the problem is wifi not working.
please fix this.

asukumar10248, 2021-05-22:
press Ctrl+Alt+t,
then type shell
then sudo edit-grub-config.
After that you should find “options=“ . Then type and make it “options=broadcom_wl”.
Finally press Ctrl+O
then press enter
then Ctrl+X.
then type sudo reboot.

Nakul-TK, 2021-05-23:
I have tried
But it is not working

Buzzcut22, 2021-07-09:
> press Ctrl+Alt+t,
> then type shell
> then sudo edit-grub-config.
> After that you should find “options=“ . Then type and make it “options=broadcom_wl”.
> Finally press Ctrl+O
> then press enter
> then Ctrl+X.
> then type sudo reboot.

sir can you post step by step screenshot for this. thank you very much

ravindrank, 2021-08-09:
Even I have the same issue.. I have added the options and it was working fine until few months ago. Over last few months, the wifi doesnt work any more. I'm sure lot of Broadcomm  users have the same. The module loads fine and I can see the drivers loaded but no wifi accesspoints show up post scanning

Ciancy28, 2021-08-18:
I'm facing the same issue on my vaio with BCM943142HM.

Nakul-TK, 2021-08-18:
> I'm facing the same issue on my vaio with BCM943142HM.

go to ctrl + alt + t {terminal} and type shell and copy all below and paste 

sudo su
mount -o remount, rw /
sudo echo "blacklist b43" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist b43legacy" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist ssb" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist bcma" >> /etc/modprobe.d/blacklist.conf
sudo echo "wl" >> /etc/modules
sudo touch /etc/modprobe.d/wl.conf
sudo echo "alias wlan0 wl" >> /etc/modprobe.d/wl.conf
cd /lib/modules/3.4.0/kernel/drivers/net/wireless/
wget http://chromium.arnoldthebat.co.uk/files/fw/lib/modules/3.4.0/kernel/drivers/net/wireless/wl.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/cfg80211.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_ccmp.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_tkip.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_wep.ko
sudo insmod /lib/modules/3.4.0/kernel/drivers/net/wireless/wl.ko
sudo depmod -a

Nakul-TK, 2021-08-18:
> Even I have the same issue.. I have added the options and it was working fine until few months ago. Over last few months, the wifi doesnt work any more. I'm sure lot of Broadcomm users have the [...]

go to ctrl + alt + t {terminal} and type shell and copy all below and paste 

sudo su
mount -o remount, rw /
sudo echo "blacklist b43" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist b43legacy" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist ssb" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist bcma" >> /etc/modprobe.d/blacklist.conf
sudo echo "wl" >> /etc/modules
sudo touch /etc/modprobe.d/wl.conf
sudo echo "alias wlan0 wl" >> /etc/modprobe.d/wl.conf
cd /lib/modules/3.4.0/kernel/drivers/net/wireless/
wget http://chromium.arnoldthebat.co.uk/files/fw/lib/modules/3.4.0/kernel/drivers/net/wireless/wl.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/cfg80211.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_ccmp.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_tkip.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_wep.ko
sudo insmod /lib/modules/3.4.0/kernel/drivers/net/wireless/wl.ko
sudo depmod -a

Ciancy28, 2021-08-18:
> go to ctrl + alt + t {terminal} and type shell and copy all below and paste
>
> sudo su
> mount -o remount, rw /
> sudo echo "blacklist b43" >> /etc/modprobe.d/blacklist.conf
> sudo echo "blacklist [...]

Doesen't seem to work.

[image]

sebanc, 2021-08-18:
Could someone try the latest brunch-unstable release ?
https://github.com/sebanc/brunch-unstable/releases

ravindrank, 2021-08-18:
> Could someone try the latest brunch-unstable release ?
> https://github.com/sebanc/brunch-unstable/releases

Thank you .. Just installed and about to reboot but wanted to mention..seeing the pv command not found error over last few releases..

chronos@localhost / $ sudo chromeos-update -f /home/user/fa3f4a33926cd292dcea3ab98c95a6a3f96d3074/Downloads/brunch_r91_unstable_20210818.tar.gz 
rootc.img
rootc.img
/usr/sbin/chromeos-update: line 56: pv: command not found
Brunch updated.

ravindrank, 2021-08-18:
No change with the unstable release.. Basically no wifi SSIDs found..

ravindrank, 2021-08-18:
> go to ctrl + alt + t {terminal} and type shell and copy all below and paste
>
> sudo su
> mount -o remount, rw /
> sudo echo "blacklist b43" >> /etc/modprobe.d/blacklist.conf
> sudo echo "blacklist [...]

How is this going to help? Asking because.. kernel version doesnt match..so those modules would never get loaded..

localhost / # uname -a
Linux localhost **4.19.139**-brunch-sebanc #1 SMP PREEMPT Sat Sep 19 09:24:56 CEST 2020 x86_64 Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz GenuineIntel GNU/Linux

sebanc, 2021-08-18:
The update did not work, there was a mandatory brunch update that you missed and now it won't let you update. You can confirm with "cat /etc/brunch_version" command that it is still the old one.

To fix it, install the "pv" package from chromebrew and try again the update:
https://github.com/skycocker/chromebrew

ravindrank, 2021-08-18:
Thanks... I was able to get pv using crew.. 

chronos@localhost / $ sudo chromeos-update -f /home/user/fa3f4a33926cd292dcea3ab98c95a6a3f96d3074/Downloads/brunch_r91_unstable_20210818.tar.gz
rootc.img
rootc.img
1.00GiB 0:00:00 [2.12GiB/s] [================================================================================================================================>] 100%            
Brunch updated.

Now version shows 91
Rebooted and still dont see the wifi listing any SSIDs..

Anything else I should try?

sebanc, 2021-08-18:
Could you attach a dmesg ? Run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your Downloads folder.

ravindrank, 2021-08-18:
Here it is ...
[dmesg1.txt](https://github.com/sebanc/brunch/files/7007728/dmesg1.txt)

sebanc, 2021-08-18:
What's the output of "sudo rfkill list all" in crosh shell ?

And also of "sudo lspci"

ravindrank, 2021-08-18:
Here it is :

chronos@localhost / $ sudo rfkill list all
Password: 
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: brcmwl-0: Wireless LAN
        Soft blocked: no
        Hard blocked: no

chronos@localhost / $ sudo lspci
00:00.0 Host bridge: Intel Corporation Device 5910 (rev 05)
00:01.0 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x16) (rev 05)
00:02.0 VGA compatible controller: Intel Corporation Device 591b (rev 04)
00:04.0 Signal processing controller: Intel Corporation Device 1903 (rev 05)
00:14.0 USB controller: Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller (rev 31)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-H Thermal subsystem (rev 31)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-H LPSS I2C Controller #0 (rev 31)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-H LPSS I2C Controller #1 (rev 31)
00:16.0 Communication controller: Intel Corporation Sunrise Point-H CSME HECI #1 (rev 31)
00:17.0 SATA controller: Intel Corporation Sunrise Point-H SATA Controller [AHCI mode] (rev 31)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #1 (rev f1)
00:1c.1 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #2 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #9 (rev f1)
00:1d.4 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #13 (rev f1)
00:1d.6 PCI bridge: Intel Corporation Sunrise Point-H PCI Express Root Port #15 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point-H LPC Controller (rev 31)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-H PMC (rev 31)
00:1f.3 Audio device: Intel Corporation Device a171 (rev 31)
00:1f.4 SMBus: Intel Corporation Sunrise Point-H SMBus (rev 31)
01:00.0 3D controller: NVIDIA Corporation Device 1c8d (rev a1)
02:00.0 Network controller: Broadcom Corporation BCM4352 802.11ac Wireless Network Adapter (rev 03)
03:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. Device 525a (rev 01)
04:00.0 Non-Volatile memory controller: Toshiba America Info Systems Device 0115 (rev 01)

sebanc, 2021-08-18:
Unfortunately the log seems pretty good (and different compared to other dmesg I had on issues with broadcom_wl driver), I don't really understand where the issue can come from...

If you have the time, could you post the output of "lsmod" in crosh shell ?

ravindrank, 2021-08-18:
:) Thanks for all your help ..here it is : 
chronos@localhost / $ lsmod
Module                  Size  Used by
vhost_vsock            20480  16
vmw_vsock_virtio_transport_common    32768  1 vhost_vsock
vsock                  40960  18 vmw_vsock_virtio_transport_common,vhost_vsock
vhost                  40960  1 vhost_vsock
wl                   6451200  0
mac80211              778240  0
dell_wmi               16384  0
intel_wmi_thunderbolt    16384  0
mxm_wmi                16384  0
xt_cgroup              16384  2
dell_laptop            24576  0
dell_smbios            16384  2 dell_wmi,dell_laptop
dell_wmi_descriptor    16384  2 dell_wmi,dell_smbios
dell_smm_hwmon         16384  0
psmouse               143360  0
tpm_tis                16384  0
wmi                    28672  5 intel_wmi_thunderbolt,dell_wmi,dell_smbios,dell_wmi_descriptor,mxm_wmi
intel_hid              20480  0
tpm_tis_core           20480  1 tpm_tis
acpi_pad               16384  0
cfg80211              724992  2 wl,mac80211
hid_logitech_hidpp     40960  0
cdc_ether              16384  0
usbnet                 32768  1 cdc_ether
btusb                  57344  0
btrtl                  16384  1 btusb
snd_hda_codec_hdmi     53248  1
hid_logitech_dj        24576  0
btbcm                  16384  1 btusb
btintel                20480  1 btusb
snd_hda_codec_realtek   122880  1
snd_hda_codec_generic    73728  1 snd_hda_codec_realtek
ledtrig_audio          16384  3 snd_hda_codec_generic,snd_hda_codec_realtek,dell_laptop
i2c_hid                28672  0
snd_hda_intel          40960  3
mei_wdt                16384  0
snd_intel_nhlt         16384  1 snd_hda_intel
mei_hdcp               16384  0
snd_hda_codec         131072  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core           81920  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
r8152                  65536  0
snd_hwdep              16384  1 snd_hda_codec
intel_lpss_pci         20480  0
mei_me                 24576  2
intel_lpss             16384  1 intel_lpss_pci
mii                    16384  2 usbnet,r8152
mei                    73728  5 mei_wdt,mei_hdcp,mei_me
idma64                 16384  0
intel_pch_thermal      16384  0
i915                 1986560  32
video                  49152  3 dell_wmi,dell_laptop,i915
intel_gtt              20480  1 i915
i2c_algo_bit           16384  1 i915
drm_kms_helper        163840  1 i915
cec                    65536  2 drm_kms_helper,i915
drm                   442368  23 drm_kms_helper,i915

sebanc, 2021-08-18:
Could you try to run in crosh shell:
"sudo modprobe lib80211"
"sudo modprobe lib80211_crypt_tkip"

And see if it helps ?

ravindrank, 2021-08-18:
The modules loaded fine...but I dont see broadcom_wl loaded..

[ 1728.143451] lib80211: common routines for IEEE802.11 drivers
[ 1728.143455] lib80211_crypt: registered algorithm 'NULL'
[ 1739.113579] lib80211_crypt: registered algorithm 'TKIP'
chronos@localhost / $ lsmod
Module                  Size  Used by
lib80211_crypt_tkip    16384  0
lib80211               16384  1 lib80211_crypt_tkip
vhost_vsock            20480  16
vmw_vsock_virtio_transport_common    32768  1 vhost_vsock
vsock                  40960  18 vmw_vsock_virtio_transport_common,vhost_vsock
vhost                  40960  1 vhost_vsock
wl                   6451200  0
mac80211              778240  0
dell_wmi               16384  0
intel_wmi_thunderbolt    16384  0
mxm_wmi                16384  0
xt_cgroup              16384  2
dell_laptop            24576  0
dell_smbios            16384  2 dell_wmi,dell_laptop
dell_wmi_descriptor    16384  2 dell_wmi,dell_smbios
dell_smm_hwmon         16384  0
psmouse               143360  0
tpm_tis                16384  0
wmi                    28672  5 intel_wmi_thunderbolt,dell_wmi,dell_smbios,dell_wmi_descriptor,mxm_wmi
intel_hid              20480  0
tpm_tis_core           20480  1 tpm_tis
acpi_pad               16384  0
cfg80211              724992  2 wl,mac80211
hid_logitech_hidpp     40960  0
cdc_ether              16384  0
usbnet                 32768  1 cdc_ether
btusb                  57344  0
btrtl                  16384  1 btusb
snd_hda_codec_hdmi     53248  1
hid_logitech_dj        24576  0
btbcm                  16384  1 btusb
btintel                20480  1 btusb
snd_hda_codec_realtek   122880  1
snd_hda_codec_generic    73728  1 snd_hda_codec_realtek
ledtrig_audio          16384  3 snd_hda_codec_generic,snd_hda_codec_realtek,dell_laptop
i2c_hid                28672  0
snd_hda_intel          40960  3
mei_wdt                16384  0
snd_intel_nhlt         16384  1 snd_hda_intel
mei_hdcp               16384  0
snd_hda_codec         131072  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core           81920  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
r8152                  65536  0
snd_hwdep              16384  1 snd_hda_codec
intel_lpss_pci         20480  0
mei_me                 24576  2
intel_lpss             16384  1 intel_lpss_pci
mii                    16384  2 usbnet,r8152
mei                    73728  5 mei_wdt,mei_hdcp,mei_me
idma64                 16384  0
intel_pch_thermal      16384  0
i915                 1986560  32
video                  49152  3 dell_wmi,dell_laptop,i915
intel_gtt              20480  1 i915
i2c_algo_bit           16384  1 i915
drm_kms_helper        163840  1 i915
cec                    65536  2 drm_kms_helper,i915
drm                   442368  23 drm_kms_helper,i915

sebanc, 2021-08-18:
Still no WiFi scan after loading the lib80211 module ?

The broadcom module name is just "wl", it looks correctly loaded and created the interfaces.

Unfortunately I am a bit lost here...

ravindrank, 2021-08-18:
ohh.. I only see below ...

chronos@localhost /lib/modules/5.4.131-brunch-sebanc $ ls
acpi_call.ko            goodix.ko         kernel             modules.builtin.alias.bin  modules.dep.bin  modules.symbols      rtl8188fu.ko  rtl8723du.ko  rtl8821cu.ko
broadcom_wl.ko          i915-oled.ko      modules.alias      modules.builtin.bin        modules.devname  modules.symbols.bin  rtl8192eu.ko  rtl8812au.ko  rtl88x2bu.ko
build                   ipts.ko           modules.alias.bin  modules.builtin.modinfo    modules.order    rtbth.ko             rtl8723bu.ko  rtl8814au.ko
drm_kms_helper-oled.ko  iwlwifi_backport  modules.builtin    modules.dep                modules.softdep  rtl8188eu.ko         rtl8723de.ko  rtl8821ce.ko
chronos@localhost /lib/modules/5.4.131-brunch-sebanc $ ls -lR | grep wl.ko
-rw-r--r--.  1 root root  7514136 Aug 18 10:49 broadcom_wl.ko

sebanc, 2021-08-18:
Yes for clarity I renamed the module file to broadcom_wl.ko but the actual module name as recognized by the kernel is still "wl"

ravindrank, 2021-08-18:
I see... just for some fun..  I installed 88, 89, 90, 91 unstable in the order and checked everytime to see if there is a change.. same behaviour.. 

I will spend some time tommorrow comparing with my Ubuntu and see what I find..

one thing though.. I havent seen the below on Ubuntu..

36.736984] wl: loading out-of-tree module taints kernel.
[   36.736992] wl: module license 'MIXED/Proprietary' taints kernel.
[   36.736995] Disabling lock debugging due to kernel taint
[   36.741119] wl 0000:02:00.0: enabling device (0000 -> 0002)
[   36.776630] wlan0: Broadcom BCM43b1 802.11 Hybrid Wireless Controller 6.30.223.271 (r587334)

sebanc, 2021-08-18:
Thanks, that would be very helpful :)

sebanc, 2021-08-18:
@ravindrank Could you try to replace "modprobe mac80211" by "modprobe cfg80211" in the file /etc/init/broadcom_wl.conf ?

And also add "blacklist mac80211" in /etc/modprobe.d/broadcom_wl.conf ?

ravindrank, 2021-08-18:
Thank you @sebanc Tried this and the module list looks fine..No change though. No listing of the wifi SSIds.. Tried to connect to known networks just be sure and got out of range..

chronos@localhost / $ lsmod
Module                  Size  Used by
vhost_vsock            20480  5
vmw_vsock_virtio_transport_common    32768  1 vhost_vsock
vsock                  40960  7 vmw_vsock_virtio_transport_common,vhost_vsock
vhost                  40960  1 vhost_vsock
wl                   6451200  0
dell_wmi               16384  0
dell_laptop            24576  0
intel_wmi_thunderbolt    16384  0
mxm_wmi                16384  0
dell_smbios            16384  2 dell_wmi,dell_laptop
xt_cgroup              16384  2
dell_wmi_descriptor    16384  2 dell_wmi,dell_smbios
dell_smm_hwmon         16384  0
psmouse               143360  0
tpm_tis                16384  0
intel_hid              20480  0
wmi                    28672  5 intel_wmi_thunderbolt,dell_wmi,dell_smbios,dell_wmi_descriptor,mxm_wmi
tpm_tis_core           20480  1 tpm_tis
acpi_pad               16384  0
cfg80211              724992  1 wl
hid_logitech_hidpp     40960  0
btusb                  57344  0
btrtl                  16384  1 btusb
btbcm                  16384  1 btusb
btintel                20480  1 btusb
hid_logitech_dj        24576  0
snd_hda_codec_hdmi     53248  1
snd_hda_codec_realtek   122880  1
snd_hda_codec_generic    73728  1 snd_hda_codec_realtek
i2c_hid                28672  0
ledtrig_audio          16384  3 snd_hda_codec_generic,snd_hda_codec_realtek,dell_laptop
snd_hda_intel          40960  2
mei_wdt                16384  0
snd_intel_nhlt         16384  1 snd_hda_intel
mei_hdcp               16384  0
snd_hda_codec         131072  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core           81920  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
r8152                  65536  0
snd_hwdep              16384  1 snd_hda_codec
intel_lpss_pci         20480  0
mei_me                 24576  2
intel_lpss             16384  1 intel_lpss_pci
mii                    16384  1 r8152
mei                    73728  5 mei_wdt,mei_hdcp,mei_me
idma64                 16384  0
intel_pch_thermal      16384  0
i915                 1986560  29
video                  49152  3 dell_wmi,dell_laptop,i915
intel_gtt              20480  1 i915
i2c_algo_bit           16384  1 i915
drm_kms_helper        163840  1 i915
cec                    65536  2 drm_kms_helper,i915
drm                   442368  19 drm_kms_helper,i915

Nakul-TK, 2021-08-19:
> > press Ctrl+Alt+t,
> > then type shell
> > then sudo edit-grub-config.
> > After that you should find “options=“ . Then type and make it “options=broadcom_wl”.
> > Finally press Ctrl+O
> > then press enter
> > then Ctrl+X.
> > then type sudo reboot.
> 
> sir can you post step by step screenshot for this. thank you very much

just copy paste in terminal

Nakul-TK, 2021-08-19:
> Doesen't seem to work.
>
> [image]

same errors and process i got but it worked for me . after all steps you should restart

Nakul-TK, 2021-08-19:
> How is this going to help? Asking because.. kernel version doesnt match..so those modules would never get loaded..
>
> localhost / # uname -a
> Linux localhost **4.19.139**-brunch-sebanc #1 SMP PREEMPT [...]

but it worked for me

Nakul-TK, 2021-08-19:
> I'm facing the same issue on my vaio with BCM943142HM.

go to ctrl + alt + t {terminal} and type shell and copy all below and paste

sudo su
mount -o remount, rw /
sudo echo "blacklist b43" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist b43legacy" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist ssb" >> /etc/modprobe.d/blacklist.conf
sudo echo "blacklist bcma" >> /etc/modprobe.d/blacklist.conf
sudo echo "wl" >> /etc/modules
sudo touch /etc/modprobe.d/wl.conf
sudo echo "alias wlan0 wl" >> /etc/modprobe.d/wl.conf
cd /lib/modules/3.4.0/kernel/drivers/net/wireless/
wget http://chromium.arnoldthebat.co.uk/files/fw/lib/modules/3.4.0/kernel/drivers/net/wireless/wl.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/cfg80211.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_ccmp.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_tkip.ko
sudo insmod /lib/modules/3.4.0/kernel/net/wireless/lib80211_crypt_wep.ko
sudo insmod /lib/modules/3.4.0/kernel/drivers/net/wireless/wl.ko
sudo depmod -a

and restart the pc

sebanc, 2021-08-19:
@ravindrank Could you try the new brunch-unstable build ?

Ciancy28, 2021-08-19:
> go to ctrl + alt + t {terminal} and type shell and copy all below and paste
>
> sudo su
> mount -o remount, rw /
> sudo echo "blacklist b43" >> /etc/modprobe.d/blacklist.conf
> sudo echo "blacklist [...]

Still doesn't work.

Nakul-TK, 2021-08-20:
> Still doesn't work.

oops . after all commands did you press enter to last command to work ?

ravindrank, 2021-08-20:
> @ravindrank Could you try the new brunch-unstable build ?
Tried the latest unstable build.. No change :( 
sorry I was out yetserday ..couldnt look into this issue

Ciancy28, 2021-08-20:
> oops . after all commands did you press enter to last command to work ?

Yes.
Otherwise i couldn't sudo reboot

sebanc, 2021-08-20:
@ravindrank No worries and thanks for trying it :)

Actually it is kind of a bad timing as I have just uploaded a new brunch-unstable build with a new potential fix (downgrading wpa_supplicant binaries), could you try it when you have the time ?

ravindrank, 2021-08-20:
Trying it now :) I'm in India ..Friday afternoon..good time :)

ravindrank, 2021-08-20:
@sebanc  Good news but there is some probably bad news as well.
Now, I'm able to connect to my know Wifi SSIDs
Bad news.. It doesn seem to list all the SSIDs in range.. Lists only known networks
Screenshot below
[image]

ravindrank, 2021-08-20:
Sorry maybe I spoke early.. But it lists one now..I'm pretty sure there are quite many SSIDs around though. So probably populating slowly!
[image]

Nakul-TK, 2021-08-20:
> Trying it now :) I'm in India ..Friday afternoon..good time :)

i am from tamil nadu (india) good afternoon

[32 more replies]

closed 2026-03-04 by github-actions[bot]