sebanc/brunch #1027: camera is not working, is there anyway to solve this problem?

wanglrebe, 2021-04-27:
love brunch
device: hp elite x2 1012 g2
recovery: rammus
cpu: i7 7600u
ram: 16gb
almost all things works fine, include pen and tablet mode
i still have a question that i can not use both of the cameras, i know front windows hello camera is not supported by linux, but my back camera is not working too.is there anyway to solve this?
thanks

SilverXanga, 2021-05-04:
My HP Spectre x360 15 (2018) also had a problem with the camera. The cors-camera process kept respawing until it was stopped. Similar to [this](https://github.com/sebanc/brunch/issues/729#issuecomment-752284100). The mentioned solution in #627 solved my problem.

Lucaacer, 2021-05-06:
> love brunch
> device: hp elite x2 1012 g2
> recovery: rammus
> cpu: i7 7600u
> ram: 16gb
> almost all things works fine, include pen and tablet mode
> i still have a question that i can not use both of the [...]

You should add no_camera_config as an option in grub.

closed 2026-03-04 by github-actions[bot]

ckomma, 2026-10-29:
have a look at https://github.com/ckomma/hp-elite-x2-1012-g2-cameras