sebanc/brunch #1029: WiFi not working on ChromeOS bcm94352z

TheEcki, 2021-04-27:
Wi-Fi does not work, Windows normally detects the device and installs drivers. On ChromeOS, it won't work.

zukuchido, 2021-04-27:
It works actually just press Ctrl+Alt+F2, write root then type sudo edit-grub-config. After that you should find “options=“ . Then type and make it “options=broadcom_wl”. Finally press Ctrl+O then press enter then Ctrl+X. Reboot your PC.

TheEcki, 2021-04-27:
Thank you very much. I will check tomorrow and let me know if it works.

TheEcki, 2021-04-28:
Wi-Fi is not searching for the network. Unfortunately it does not work.

asukumar10248, 2021-05-22:
press Ctrl+Alt+t,
then type shell
then sudo edit-grub-config.
After that you should find “options=“ . Then type and make it “options=broadcom_wl”.
Finally press Ctrl+O 
then press enter 
then Ctrl+X. 
then type sudo reboot.

asukumar10248, 2021-05-22:
It works actually just press Ctrl+Alt+t, write shell then type sudo edit-grub-config. After that you should find “options=“ . Then type and make it “options=broadcom_wl”. Finally press Ctrl+O then press enter then Ctrl+X.then sudo reboot

ravindrank, 2021-08-09:
Even I have the same issue.. I have added the options and it was working fine until few months ago. Over last few months, the wifi doesnt work any more. I'm sure lot of Broadcomm  users have the same. The module loads fine and I can see the drivers loaded but no wifi accesspoints show up post scanning