sebanc/brunch #1033: grub not boot on a mbr computer

Agazart, 2021-04-29:
Hello, i installed chrome os with brunch and mbr support but grub not boot.
My computer detects something in partition 12 but it does not want to launch it
 if anyone had the same problem, thanks.

Sayor0630, 2021-05-03:
You can't dual-boot ChromeOS on Legacy BIOS. You have to install ChromeOS on the whole Machine. If you want to dual-boot, you must have UEFI BIOS with a GPT partition table. Thank you.

Agazart, 2021-05-03:
Hi, i try just install ChromeOs and grub don't boot on my machine but i have a compatible cpu.

kerta1n, 2021-05-05:
You need to follow [these](https://github.com/sebanc/brunch#limited-support-for-mbrbios-devices) instructions. Compatibility is very limited, however.

Sayor0630, 2021-05-05:
> Hi, i try just install ChromeOs and grub don't boot on my machine but i have a compatible cpu.

Can you give me every details about your PC and installation? I want to help you.

Agazart, 2021-05-05:
> > Hi, i try just install ChromeOs and grub don't boot on my machine but i have a compatible cpu.
> 
> Can you give me every details about your PC and installation? I want to help you.

My computer is Asus laptop, his reference is R505CB-XO450H and my cpu is a Intel Core i3-3217U. I tried to install brunch r89 with the mbr support and i used chrome Os Sammus 87. On linux everything is installed correctly but grub not boot, i boot directly in the bios. 
Thanks

Agazart, 2021-05-05:
and i try this install on kali linux live.

Sayor0630, 2021-05-05:
Are you dual booting? You have mbr partition. Right?

Agazart, 2021-05-05:
Yes, I have an mbr partition but I am not in dual boot

Sayor0630, 2021-05-05:
Hmm. You are using Kali Linux grub. But you didn't install kali linux. That's why your ChromeOS not booting. I am copying the reference link for installing ChromeOS. If must have a pendrive of at least 16GB. -------

https://github.com/sebanc/brunch#limited-support-for-mbrbios-devices

then follow this--

https://github.com/sebanc/brunch#install-chromeos-on-a-usb-flash-drive--sd-card--hdd-full-disk-install--single-boot

Hope this will work.

Sayor0630, 2021-05-05:
You must read this sincerely--
https://github.com/sebanc/brunch#brunch-framework

Sayor0630, 2021-05-05:
You can see this video.
https://www.youtube.com/watch?v=UElQAh5SCmo

Agazart, 2021-05-05:
Thanks, i will try exactly like in the video. i will get back to you

Agazart, 2021-05-05:
I installed hrome os by following the video with brunch 89 and mbr support but when I run my computer, this happens: 
https://youtu.be/Dc78EUPeTk0
This time I install it with a linux installer and not a live linux

Sayor0630, 2021-05-06:
Can you give me a picture of what it says? It can help up in this case.

Agazart, 2021-05-06:
Nothing happens on the Grub side, I boot in the BIOS as if I did not have a disk but when I create a new BIOS option, the computer knows that there is something in partition 12.

Sayor0630, 2021-05-06:
You don't need to create BIOS option. There is no need. You can see this video. See if it works.
https://www.youtube.com/watch?v=UElQAh5SCmo. I think this will work.

Sayor0630, 2021-05-06:
I will make a bootable img. Then I will upload it in Google drive. Then I will tell you what to do.

Sayor0630, 2021-05-06:
Do you have a pen drive of at least 16GB?

Agazart, 2021-05-06:
I followed step by step the video and I have the same problem 🤔

Sayor0630, 2021-05-06:
Can you make a short video of loading?

Sayor0630, 2021-05-06:
Do you have dual GPUs? Dual GPU can cause this problem.

Agazart, 2021-05-06:
Yes I have a Nvidia 750m and an Intel graphics.

Sayor0630, 2021-05-06:
Ooh. It caused the problem. I disable a GPU.  I forgot the command. I will tell you after I remember.

Sayor0630, 2021-05-06:
Does your motherboard support UEFI?

Agazart, 2021-05-06:
I think that is just legacy.

Agazart, 2021-05-06:
and support uefi, my bad .

Sayor0630, 2021-05-07:
UEFI is supported? Then use UEFI. Convert your HDD in GPT. Disable your Nvidia 750. I hope it will work.

Agazart, 2021-05-07:
Ok I try this but how disable my graphic card ?

Sayor0630, 2021-05-07:
I really forgot ChromeOS grub command.

Sayor0630, 2021-05-07:
I created an Issue. Hope anyone will reply that.

Agazart, 2021-05-08:
https://github.com/sebanc/brunch/issues/357 

i find this for disable dualgpu

Sayor0630, 2021-05-09:
Did you install ChromeOS?

Agazart, 2021-05-10:
Hey, today, i reinstall chrome os and it's WORK. Thank you so much.
Finally I installed chrome os, on linux with no mbr support and I added a boot option with this path "\ EFI \ BOOT \ GRUBX64.EFI" and it worked directly.

Thank you.
Constan

closed 2021-05-12 by Agazart