sebanc/brunch #1039: Cannot boot on mipad2 which has Intel Atom X5-Z8500

lelemka0, 2021-05-03:
I am trying to install chromeos on my mipad2 which has Intel Atom X5-Z8500.
It has a weird graphics card that linux cannot boot without kernel option "nomodeset".
On mipad2, other linux distributions such as ubuntu can be booted and enter the desktop with "nomodeset".
But if "nomodeset" is not set, the screen will be black after the Grub.

I followed the instructions and tried to boot chromeos using usb media. By default, the kernel stopped at 

> fb0: switching to inteldrmfb from efi vga

And after I add "nomodeset" in kernel options, the last line of the kernel log is 

> init: ui main process terminated with status 2

This content appears many times.

I tried to switch kernel to 4.19 and 5.10, and also tried three versions of r87, r88 and r89, but to no avail.

I found a similar question on reddit, 
https://www.reddit.com/r/Brunchbook/comments/lrve8p/gpu_error_when_install_brunch_r88_on_mipad_2/
The difference is that the questioner seems to be able to boot correctly using nomodeset.

I am sorry for my bad English.

2907405506, 2024-02-19:
I also have this problem, have you solved it?😭😭😭

lelemka0, 2024-02-23:
> I also have this problem, have you solved it?😭😭😭

no. in fact, there are also many problems running ubuntu on mipad2. i have tried for a long time and there is no solution at all.
i have stopped trying to run any os other than windows.

Qs315490, 2025-09-27:
Hello, I have reviewed the repository link you provided. According to the issue you described, I suggest you try applying the patch file [`latte-i915-fix.patch`](https://github.com/Qs315490/brunch-unstable-latte/raw/refs/heads/continuous-integration/kernel-patches/6.12/mipad2/latte-i915-fix.patch) to your Chrome OS kernel source code. Here are the specific steps:

1. Download the patch file `latte-i915-fix.patch`.

2. Use the `git` command-line tool to apply the patch file to your Chrome OS kernel source code. Here is an example command:

```bash

git apply path/to/latte-i915-fix.patch

```

3. After applying the patch, rebuild the kernel image.

4. Burn the new kernel image to a USB drive and try using it to install Chrome OS.

If the issue still persists, please provide more detailed information so that I can further assist you in resolving the problem. Good luck!