sebanc/brunch #1047: battery is draining so fast ..high cpu usage and heating the motherboard

divya789vidya, 2021-05-10:
[image]
hlo sir its my hp 15 da0326tu laptop and its processor is i3 7100u processor  with 12gb ram  ... i have recently installed the chrome os in ssd partition 
I was expecting fast performance  , i got the great performance but now the heating is the problem whole laptop is heating badly due to this fan is working in  high speed and battery life is also getting lowered .... 

i this may it was due to hyperthreading but after removing the hyperthreading also its been heating badly 
 please help me for this problem

divya789vidya, 2021-05-10:
but for another os just as windows, it's working great no that much heating and battery also work well ...

asukumar10248, 2021-05-10:
i am also faceing the same issue i have i3 8130u and 16 gb ram and 1tb + 240gb m.2 ssd

divya789vidya, 2021-05-10:
> i am also faceing the same issue i have i3 8130u and 16 gb ram and 1tb + 240gb m.2 ssd

I was expecting the battery would drain slowly than windows but it had been halved

nadlax, 2021-05-20:
Install Fyde OS, it's more optimized

SoFLazerTag, 2021-05-24:
> i am also faceing the same issue i have i3 8130u and 16 gb ram and 1tb + 240gb m.2 ssd

Have you tried using the nami recovery image? My previous HP x360 14 chromebook was the model with the i3-8130u and was a nami board.  Maybe that image will better support your hardware.  Just a thought.

closed 2021-05-28 by divya789vidya

shadab2097, 2022-06-27:
> > > i am also faceing the same issue i have i3 8130u and 16 gb ram and 1tb + 240gb m.2 ssd
> > 
> > 
> > Have you tried using the nami recovery image? My previous HP x360 14 chromebook was the model with the i3-8130u and was a nami board. Maybe that image will better support your hardware. Just a thought.
> 
> ok i will try this .....but mostly problem has been solved it was due to media container

How did you solve the problem. I have same issue. Please help me.